bigtreetech/BIGTREETECH-SKR-V1.3 #501: SKR 1.4 Turbo + MKS TFT32 configuration issue (Marlin 2.0)

lexcassi, 2021-02-06:
Hi guys!

I'm trying to attach the MKS TFT32 touch display to the SKR 1.4 Turbo via the TFT module, but without success so far.

Some lines about my Marlin config here below :

**#define SERIAL_PORT -1
#define SERIAL_PORT_2 0 
#define BAUDRATE 115200**

I' ve already flashed the firmware into the MKS TFT32 and some lines about the mks_config.txt (TFT28_32_v3.0.6 Release file) are here below :

#mainboard firmware setting(marlin:1; repetier:2; smoothie:3)
**>cfg_firmware_type:1**
#machine setting (Normal:1; Delta:2)
**>cfg_machine_type:1**
#baud rate (9600:1; 57600:2; 115200:3; 250000:4)
**>cfg_baud_rate:115200**

This is my wiring (I've also tryed to switch Rx and Tx) :

[image]

I've another LCD attached via EXP1 and EXP2 correctly working (Marlin easily configured as follow) : 

**#define REPRAP_DISCOUNT_SMART_CONTROLLER**

As the person who sold me the printer had these two display working correctly attached to his old MKS Gen L v1.0 with Marlin 1.1.8, I've also tried enabling the following as he did:

**#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER**

But this raised an error in the SanityCheck.h ("Make sure only one display is enabled"), so I commented out this particular check (lines 2286 to 2341)

Any hints about what I'm doing wrong/right? I'm walking around blind (probably due my own inexperience...)

Thank you so much for your really appreciated help!

MKS TFT32 full schematics :
https://github.com/makerbase-mks/MKS-TFT-Hardware/blob/master/MKS%20TFT32/MKS%20TFT32_L%20V3.x/MKS%20TFT32_L%20V3.1_001%20SCH.pdf

radek8, 2021-02-07:
Do you have the right speed set?

 #baud rate (9600:1; 57600:2; 115200:3; 250000:4)
 **>cfg_baud_rate:3**

 No

 #baud rate (9600: 1; 57600: 2; 115200 : 3; 250000: 4)
 **> cfg_baud_rate: 115200**

lexcassi, 2021-02-07:
Thank you so much, it was really a silly mistake, now it works perfectly!!!

closed 2021-02-07 by lexcassi

Ducktr00per, 2022-06-02:
do you need to activate something in marlin 2.0 like
// 320x240, 3.2", FSMC Display From MKS
// Usually paired with MKS Robin Nano V1.2
//
#define MKS_ROBIN_TFT32

and which tft ui ?

K2000sv, 2022-10-24:
@lexcassi , I hope you are doing well. I have a silly question but wanted to confirm, in the wiring the TX of the TFT card goes to RX0 in the control board SKR and RX of TFT to TX0 in the SKR, or it was just a mistake?
Thank you in advance for your help on this matter.

lexcassi, 2022-10-28:
@K2000sv Hi, RX and TX are for receiving and transmitting data, usually the RX from one device should go to the TX of the other, and vice-versa. 
Hope that helped, have a good one!

K2000sv, 2022-11-02:
Makes sense, thank you.