bigtreetech/BIGTREETECH-SKR-V1.3 #502: #define serial port 3

davbcr, 2021-02-10:
I'm trying to get Marlin2 up and running on a skr 1.4 turbo with tft35 and esp01s wifi module (on the board) bit keeping usb connection working too.
So I #define serial port to -1 as usual, serial port 2 to 0 for tft screen and serial port 3 to 3 that's the esp01s pins referred 4.28 and 4.29.
Compiling goes well but does it work on the board?
I mean will I be able to access the board both from touch screen and wifi module and usb,or Marlin has a maximum of 2 serial connections active at a time?

davbcr, 2021-02-10:
It works!
[https://www.facebook.com/messenger_media/?thread_id=1525689265&attachment_id=123573476326458&message_id=mid.%24cAAAAAAkGDbB9u6sgXF3imXS2aMII]

closed 2021-02-10 by davbcr

Cheo7811, 2021-02-10:
What was the solution? your attachment comes as a link that doesn't work

davbcr, 2021-02-10:
I just simply add the 3rd serial port with
#define serial_port_3 0

Cheo7811, 2021-02-11:
> I just simply add the 3rd serial port with
> #define serial_port_3 0

Sorry, I am still missing something, what values do you use for the other two ports?
#define serial_port -1
#define serial_port_2 ?
#define serial_port_3 0

davbcr, 2021-02-11:
> > I just simply add the 3rd serial port with
> > #define serial_port_3 0
> 
> Sorry, I am still missing something, what values do you use for the other two ports?
> #define serial_port -1
> #define serial_port_2 ?
> #define serial_port_3 0

Here they are.
#define serial_port -1
#define serial_port_2 3
#define serial_port_3 0