bigtreetech/BIGTREETECH-SKR-V1.3 #505: Shifting BL Touch readings and Z Offset on SKR 1.4 Turbo

NastyEwok, 2021-02-16:
I seem to be having very strange issues with my SKR 1.4 Turbo 4 x TMC2209 v1.2 drivers on UART.

I install the Firmware (Latest Marlin Bugfix, followed Gadget Angels Guide) reset the eeprom and return to defaults. Set the Z Offset with a 0.8 feeler gauge, and run BLTouch on a 7x7 pattern, 2 reads at each point.

First issue, My Z Offset seems to constantly be shifting, set it to -2.0, then check it again and it is -2.3, rehome a few times and check, start a job and just about scraping the bed, check again and it is back to -2.0, this has been going on for ages.

Have just ran 2 bed levels
[Marlin.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/5990943/Marlin.zip)
[Marlin.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/5990947/Marlin.zip)

 back to back, bed heated to 55 and 2 completely different sets of results as below.

Have I got a faulty board here, or FW config issues...its driving me insane.

Can someone help please

0 1 2 3 4 5 6

0 -1.456 -1.369 -1.089 -0.919 -0.698 -0.515 -0.461

1 -1.331 -1.265 -1.014 -0.852 -0.626 -0.453 -0.410

2 -0.957 -0.914 -0.683 -0.556 -0.353 -0.202 -0.187

3 -0.753 -0.728 -0.512 -0.402 -0.208 -0.066 -0.059

4 -0.419 -0.424 -0.238 -0.154 +0.007 +0.118 +0.086

5 -0.160 -0.174 +0.013 +0.090 +0.263 +0.366 +0.330

6 +0.156 +0.120 +0.279 +0.336 +0.465 +0.520 +0.444

0 -0.747 -0.667 -0.384 -0.216 -0.003 +0.194 +0.236

1 -0.621 -0.556 -0.306 -0.142 +0.073 +0.260 +0.290

2 -0.242 -0.202 +0.028 +0.159 +0.352 +0.504 +0.521

3 -0.035 -0.015 +0.199 +0.322 +0.502 +0.654 +0.658

4 +0.297 +0.292 +0.479 +0.570 +0.715 +0.836 +0.802

5 +0.560 +0.540 +0.723 +0.814 +0.969 +1.081 +1.040

6 +0.879 +0.836 +0.993 +1.049 +1.166 +1.238 +1.157

What the heck is going on?

radek8, 2021-02-17:
Uncomment:
#define BLTOUCH_DELAY 500
#define MONITOR_DRIVER_STATUS
#define HYBRID_THRESHOLD
#define TMC_DEBUG

What will Marlin answer for your M122 command?

I'm not sure about your value
#define Z_MIN_PROBE_ENDSTOP_INVERTING
I think it should be set to "true"

NastyEwok, 2021-02-18:
[Marlin.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/5999773/Marlin.zip)

This is really getting silly now, No matter what I try I cannot get a first layer to go down at all.

I have now rebuilt my printer to make sure it was not something mechanical, and tried a different preconfigured, latest release 2.0.7.2 firmware from here.

https://oddity.oddineers.co.uk/2020/10/28/marlin-2-0-7-2-ender-3-pro-btt-skr-1-4-turbo-with-abl-using-bltouch-v3-1/

I have installed the firmware, Ran M502 and M500, then G28 to autohome, set the Z offset so a .8 feeler just fits under, run a 7x7 2 point bilinear mesh and then when I start the print it seems to have completely lost the bed level data.

Can anyone assist with this please, It is driving me insane

Here is the M122 report

Bilinear Leveling Grid:
      0      1      2      3      4      5      6
 0 -0.182 -0.102 -0.051 -0.093 +0.035 -0.030 +0.013
 1 -0.249 -0.127 -0.052 -0.058 +0.099 +0.067 +0.143
 2 -0.149 -0.068 -0.018 -0.063 +0.058 -0.013 +0.028
 3 -0.230 -0.118 -0.048 -0.075 +0.069 +0.022 +0.088
 4 -0.086 -0.014 +0.022 -0.049 +0.061 -0.028 +0.001
 5 -0.140 -0.058 -0.007 -0.056 +0.064 -0.003 +0.046
 6 +0.041 +0.075 +0.099 +0.007 +0.101 -0.008 +0.007
echo:busy: processing
echo:Settings Stored (818 bytes; crc 2745)
echo:Settings Stored (818 bytes; crc 2745)
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:Settings Stored (818 bytes; crc 53388)
echo:Settings Stored (818 bytes; crc 54109)
echo:enqueueing "M23 ce3pro~1.gco"
echo:enqueueing "M24"
echo:Now fresh file: ce3pro~1.gco
File opened: ce3pro~1.gco Size: 34609
File selected
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
>>> M122
SENDING:M122
axis:pwm_scale/curr_scale/mech_load|flags|warncount
		X	Y	Z	E
Address		0	0	0	0
Enabled		false	false	false	false
Set current	600	600	600	500
RMS current	581	581	581	489
MAX current	819	819	819	689
Run current	18/31	18/31	18/31	15/31
Hold current	9/31	9/31	9/31	7/31
CS actual	9/31	9/31	9/31	7/31
PWM scale
vsense		1=.18	1=.18	1=.18	1=.18
stealthChop	true	true	true	true
msteps		32	32	32	32
tstep		max	max	max	max
PWM thresh.
[mm/s]
OT prewarn	false	false	false	false
triggered
 OTP		false	false	false	false
pwm scale sum	29	29	33	11
pwm scale auto	1	1	0	0
pwm offset auto	91	90	0	44
pwm grad auto	28	26	0	27
off time	4	4	4	4
blank time	24	24	24	24
hysteresis
 -end		2	2	2	2
 -start		1	1	1	1
Stallguard thrs	0	0	0	0
uStep count	84	556	140	516
DRVSTATUS	X	Y	Z	E
sg_result	84	74	80	2
stst
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
		X	0xC0:09:00:00
		Y	0xC0:09:00:00
		Z	0xC0:09:00:00
		E	0xC0:07:00:00
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... OK
X:29/9/-|TI|	Y:29/9/-|TI|	Z:33/9/-|TI|	E:11/7/-|TI|
X:29/9/-|TI|	Y:29/9/-|TI|	Z:33/9/-|TI|	E:11/7/-|TI|
X:29/9/-|TI|	Y:29/9/-|TI|	Z:33/9/-|TI|	E:11/7/-|TI|
X:29/9/-|TI|	Y:29/9/-|TI|	Z:33/9/-|TI|	E:11/7/-|TI|
X:29/9/-|TI|	Y:29/9/-|TI|	Z:33/9/-|TI|	E:11/7/-|TI|
X:29/9/-|TI|	Y:29/9/-|TI|	Z:33/9/-|TI|	E:11/7/-|TI|
X:29/9/-|TI|	Y:29/9/-|TI|	Z:33/9/-|TI|	E:11/7/-|TI|
X:29/9/-|TI|	Y:29/9/-|TI|	Z:33/9/-|TI|	E:11/7/-|TI|
X:29/9/-|TI|	Y:29/9/-|TI|	Z:33/9/-|TI|	E:11/7/-|TI|
X:29/9/-|TI|	Y:29/9/-|TI|	Z:33/9/-|TI|	E:11/7/-|TI|
X:29/9/-|TI|	Y:29/9/-|TI|	Z:33/9/-|TI|	E:11/7/-|TI|
X:29/9/-|TI|	Y:29/9/-|TI|	Z:33/9/-|TI|	E:11/7/-|TI|
X:29/9/-|TI|	Y:29/9/-|TI|	Z:33/9/-|TI|	E:11/7/-|TI|
X:29/9/-|TI|	Y:29/9/-|TI|	Z:33/9/-|TI|	E:11/7/-|TI|
X:29/9/-|TI|	Y:29/9/-|TI|	Z:33/9/-|TI|	E:11/7/-|TI|
X:29/9/-|TI|	Y:29/9/-|TI|	Z:33/9/-|TI|	E:11/7/-|TI|
X:29/9/-|TI|	Y:29/9/-|TI|	Z:33/9/-|TI|	E:11/7/-|TI|
X:29/9/-|TI|	Y:29/9/-|TI|	Z:33/9/-|TI|	E:11/7/-|TI|
X:29/9/-|TI|	Y:29/9/-|TI|	Z:33/9/-|TI|	E:11/7/-|TI|
X:29/9/-|TI|	Y:29/9/-|TI|	Z:33/9/-|TI|	E:11/7/-|TI|
X:29/9/-|TI|	Y:29/9/-|TI|	Z:33/9/-|TI|	E:11/7/-|TI|
X:29/9/-|TI|	Y:29/9/-|TI|	Z:33/9/-|TI|	E:11/7/-|TI|
X:29/9/-|TI|	Y:29/9/-|TI|	Z:33/9/-|TI|	E:11/7/-|TI|
X:29/9/-|TI|	Y:29/9/-|TI|	Z:33/9/-|TI|	E:11/7/-|TI|
X:29/9/-|TI|	Y:29/9/-|TI|	Z:33/9/-|TI|	E:11/7/-|TI|
X:29/9/-|TI|	Y:29/9/-|TI|	Z:33/9/-|TI|	E:11/7/-|TI|
X:29/9/-|TI|	Y:29/9/-|TI|	Z:33/9/-|TI|	E:11/7/-|TI|
X:29/9/-|TI|	Y:29/9/-|TI|	Z:33/9/-|TI|	E:11/7/-|TI|
X:29/9/-|TI|	Y:29/9/-|TI|	Z:33/9/-|TI|	E:11/7/-|TI|
X:29/9/-|TI|	Y:29/9/-|TI|	Z:33/9/-|TI|	E:11/7/-|TI|
X:29/9/-|TI|	Y:29/9/-|TI|	Z:33/9/-|TI|	E:11/7/-|TI|
X:29/9/-|TI|	Y:29/9/-|TI|	Z:33/9/-|TI|	E:11/7/-|TI|
X:29/9/-|TI|	Y:29/9/-|TI|	Z:33/9/-|TI|	E:11/7/-|TI|

FunnyGermanGuy, 2021-03-08:
Got the very same issue. But in my case with an capacitiv sensor. Ran a g28 and the a g29 approx all 20 minutes. After 90 minutes the level measurements went from -0.012mm to 6.423mm. Needless to say thats not a usable situation.

FunnyGermanGuy, 2021-03-08:
Okay I managed to find a solution for my issue. Still got a little temp drift, but that maybe my sensor. I had a problem with my endstop connection. How did you connect the blt touch and what is your endstop configuration, i mean in hardware

zed-maker, 2021-08-20:
Have you tried setting your initial z offset gap to less than 0.8mm? That seems too big to get bed adhesion.  I’ve always used 0.1 to 0.15mm. I would be surprised if you get any adhesion at 0.8mm. Unless you mean 0.08mm that is.  I found bed adhesion to be hugely improved when I upgraded from Marlin 1.1.9 to 2.0bugfix . Variations in BL Touch readings that I have come across were from the mounting nuts and bolts on the BL touch to the hot end working themselves loose. Check all your screws, bolts and mechanicals on the entire printer to make sure they are all still tight. One loose bolt is all it takes to stuff things up. Good luck and I hope you work it out.