bigtreetech/BIGTREETECH-SKR-V1.3 #507: How can I use a stepper extender on the SKR 1.4 using TMC2208 in UART mode ?

rafik24, 2021-03-04:
Hi,
I am using an SKR 1.4 Turbo and I am adding 2 extra extruders stepper motors, for the first I am using a TMC2208 in UART mode on the E1 connector.
However for the third motor I am using a TMC2208 on an expender boar on exp1 pins for STEPS/DIR/EN.

I there a way to connect the PDN(UART) pin to the SKR 1.4 pins (Could maybe use the RX0 and TX0 pins of the TFT connector ? If yes what are these pins numbers / P1_XX ??)

Could you please help ?

Thanks in advance,
Rafik

radek8, 2021-03-04:
[image]

radek8, 2021-03-04:
[image]

radek8, 2021-03-04:
[image]

rafik24, 2021-03-04:
Thank you so much @radek8 for your prompt response pointing me in the right direction.

Many thanks !!!

rafik24, 2021-03-06:
Hi,

Received the expander board and configured the pins in klipper and all is working great.

Many thanks !

closed 2021-03-06 by rafik24