bigtreetech/BIGTREETECH-SKR-V1.3 #508: SKR 1.4 turbo E0 driver = magic smoke

Balduc, 2021-03-04:
Hi everyone. I've order an SKR 1.4 turbo and follow instructions of _**Chris Basement**_ and _**Teaching Tech**_ YouTube channel to compile and upload MARLIN 2.0.x. The board came with the TMC2209 driver, so I've enable UART mode, use sensorless homing for X and Y and cut the sensor pin of the Z, E0 and E1 on each other TMC2209. 

Everything was perfect was working flawlessly. After a month the board kind of... died. After a quick troubleshoot, I'm unable to say what happened, but i was unable to communicate via serial(USB), and the reprap 12864 LCD full graphic controller was receiving the sufficient voltage, but nothing was happening. The board didn't want to boot. But before that, it was totally fine.

So, I've re-order a new one, re-compile Marlin (the same code as the other one), put on the drivers, turn it on to upload Marlin on it, and... snap crackle pop, MAGIC SMOKE from the E0 driver. The little chip underneath the driver on the E0 burned. I was like "ok maybe the driver was not good, and the other board died because of that driver". 

So I've check the jumpers, and my code, put another tmc2209, and power it up. Hurray, everything works... for like 1 minutes or so. The E0 driver (not the same one, but the same emplacement) snapped crackled and popped with magic smoke again. I've re-check my compilation, it was fine, for me... Now I'm afraid that I may have change the code by mistake.

What i can do to avoid that kind of issue in the future with any kind of boards? Or to go further: is there a setting that can lead to that kind of issue? Nothing was connected (bed, hot ends, steppers or sensors). Only 24 (23.97) volt from a bench power supply, and a LCD 12864 smart graphic controller. The current draw was 0.64 amps and ramp up just before the burn of the drivers, but didn't see it. I've just concentrated myself to unplugged everything. Why on first boot the driver burned immediately, and on the second boot, it took a minute? A capacitor issue? Note that the drivers was working fine on the other boards.

I will re-code everything for the beginning. I just hope it was not my code, but just in case, I need your opinion. Thank you for your time. So here's my MARLIN 2.0.x source that I've compiled with VS / plateformIO:

https://drive.google.com/file/d/11_SPINRwsQfv3pVEPf4aSFKXqCz7qdgO/view?usp=sharing

radek8, 2021-03-05:
I can't open your configuration file.
I can't imagine a Marlin configuration that would cause such behavior.
The first TMC controller could be defective and by burning it contaminated the SKR board with combustion products that are conductive and subsequently the second TMC controller also burned down.

Balduc, 2021-03-05:
After a lot of posting in different forum, it might be the board itself. I've noticed that purchase those are like the lottery. I hope the next one will not have any quality issue.

https://drive.google.com/file/d/11_SPINRwsQfv3pVEPf4aSFKXqCz7qdgO/view?usp=drivesdk

radek8, 2021-03-06:
I own one SKR1.4 board and one SKR1.4 Turbo.
I have TMC2208 and TMC2209 drivers,
everything works flawlessly.
Only on the SKR1.4 board did I not clean the solder flux when soldering the connector on the print head. soldering flux propijiola temperature sensor with 24V pin,
and the pin on the processor measuring the temperature was destroyed. But that was my fault.
Fortunately, there was another pin to which it could be connected.

In my experience, if a component burns in the same way as for you, part of the metal conductors will evaporate, which will then condense in the immediate vicinity. This will create a conductive surface on everything close to the point of ignition. Even if you manage to remove the combustion products, it is quite likely that some pins on the processor leading to the driver will be destroyed.
The processor pins are sensitive to voltages above 3.3V
The driver is powered by 24V If a short circuit occurs ....