bigtreetech/BIGTREETECH-SKR-V1.3 #511: BTT-SKR-V1.4 2209 SENSORLESS BLTOUCH NO VOLTAGE

reblacklaf, 2021-03-15:
I have just finished a rebuild of my printer. I am using the V1.4 Turbo from BTT with 2209 sensorless settings. I have got things compiled and have run into an issue. I have an early Bltouch 3.0 (got in early 2019). Got x-y running but when it tries to home the zed it just keeps going. The touch never triggers after hitting the bed. Voltage diff is only about 0.2 from the time above the bed until the hotend crashes into the bed and the red led never goes out. Test run on the Bltouch says ok. Running diag says it's ok but it never triggers. Is it possible the Bltouch is bad.

Reason for the upgrade was the usb port on my Re-Arm went out and I couldn't communicate with the board from my computer. I also upgraded to the 2209's from BTT at the same time. Per some video's online I cut the diagnostic pin for the Zed axis and the extruder 2209's.

Any help would be greatly appreciated. This is my first attempt at redoing the firmware and it has taken me 4 days to get this far.

radek8, 2021-03-15:
Upload connection photo and configuration files Marlin.

reblacklaf, 2021-03-15:
Don't have an image account but on the 2 plug from the exp socket and going up is Black, White. The 3 plug from the 2 plug up is Orange, Red and Brown.

[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/6145104/Configuration.zip)
[Pics.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/6145109/Pics.zip)

Randy

radek8, 2021-03-16:
I wanted a photo of the connection, not a photo of the connectors

reblacklaf, 2021-03-16:
Sorry for confusion on my part. But what exactly do you want a photo of. At 74 and having ADD I am sometimes a bit slow.

Randy

radek8, 2021-03-16:
Take a SKR board with the connectors connected.
Take a picture of BLTouch to see where the wire is connected.

radek8, 2021-03-16:
I also have SKR1,4 with BLTouch.
You set:
#define Z_MIN_PROBE_ENDSTOP_INVERTING true
#define BLTOUCH_DELAY 500
#define BABYSTEPPING

If you use the "Probe" connector set:
// # define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define USE_PROBE_FOR_Z_HOMING

reblacklaf, 2021-03-16:
I hope this is what you want.
[BL_Cont.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/6151466/BL_Cont.zip)

Randy

radek8, 2021-03-16:
[image]

radek8, 2021-03-16:
Do you have a swapped black and white wire in the connector?

radek8, 2021-03-16:
You write that you lack a 5V power supply.
Where are you missing?

reblacklaf, 2021-03-16:
Nope it's correct. I should say that the pin will retract but it
continually drops until it shuts down and kills the printer.

Randy

reblacklaf, 2021-03-16:
Yep, continuity confirms that is the hook up I am using. I just used the
wires that the local shop had used when they set up the Re-Arm. I'm
beginning to think the Bltouch isn't working. As I said I can place the zed
above the bed and deploy the pin. Then I touch the pin, it doesn't retract
nor does the red LED go out.

Randy

reblacklaf, 2021-03-16:
It has 5v going to it.

Randy

radek8, 2021-03-16:
> It has 5v going to it. Randy

I do not understand you

radek8, 2021-03-16:
Was green originally black?

[image]

reblacklaf, 2021-03-16:
Labeled incorrectly. Checked continuity didn't change the label.

Randy

radek8, 2021-03-16:
That's why I wanted the photos to know that you have it connected correctly.
You can't see anything at all from your photos! And even what can be seen is not true.

radek8, 2021-03-16:
So is it connected according to the diagram?
Did you adjust Marlin as recommended above?

reblacklaf, 2021-03-16:
Yep, I followed video's from Chris's Basement and TheTechChief for setup. I
checked those and they are the same in my config files.

Randy

radek8, 2021-03-16:
We probably don't understand each other. I can't help you unless you can provide me with basic information.
Maybe someone else will help you.
Happy print.

reblacklaf, 2021-03-16:
Thank you for your help.

Randy

nathandtrone, 2021-08-16:
I have a BTT-SKR-V1.4 Turbo with 2209 drivers, BLTouch, and Heated Bed. I am researching how to setup this board(I'm replacing a dead board). I've seen many comments on Marlin being bugged with BLTouch I'm not sure. Did you happen to find a solution to this? @reblacklaf, would you like me to attempt a build with your configs? (Also, do you recall if you used Marlins repository or the firmware from this repository?)

I'm trying to do sensorless homing for X & Y, but use BLTouch for Z. I would like to use the mag sensors for XYZ Max, but I'm still learning how the firmware works. I've read that some people are blowing boards up by connecting what I want to, so I'm hesitant to try it. Any information you have would be helpful, and hopefully you had found a solution, if not, I'm down to work the issue with you.

reblacklaf, 2021-08-16:
I will send you a copy of the config files that I used for you to look at
later this week. Not at the system at the moment.

Randy