bigtreetech/BIGTREETECH-SKR-V1.3 #515: SKR 1.4 Turbo + TMC2209 on CR-10S

vitucciog, 2021-03-23:
Hello,
I've compiled Marlin bugfix 2.0.x with the configuration files proposed for BigTreeTech SKR 1.4 Turbo TMC2209 on my CR-10S.
The CR-10S has two Z axis and I connected them to the same driver using the ZMA and ZMB connectors on the board, that connect wires in parallel.

In Configuration_adv.h the following stepper current are defined:
    #define X_CURRENT       650
    #define Y_CURRENT       650
    #define Z_CURRENT       580
    #define E0_CURRENT       650

Using those values all steppers goes well but the Z axis can't move (it goes down with difficulty, but not up). I've succeeded to let them move correctly by increasing the Z_current to values around 800mA.

Now, I don't want to adopt a random value so I'm trying to understand how to calculate it.

As far as I could understand, for the TMC2209 it's just needed to define the stepper current based on the maximum current of the steppers taking also a 10% of margin to stay safe from overheating (sometimes 20% is suggested to stay even safer).

On CR-10S the X, Y and E0 steppers are Creality3D 42-40 (MS17HD2P410A). Their rated current is 1A,
The Z1 and Z2 stepper are Creality3D 42-34. I didn't want to dismantle it to find code, but found on google the rated current is 0.84A.

So,
- X, Y and E0 current should be (1000mA-10%)/1,41 = 638mA which is about the 650mA proposed by the standard configuration
- Z current for one motor should be (840mA-10%)/1,41 = 536mA which is also close to the 580mA proposed. But in this case two motors are connected in parallel, so the current should be doubled. Is that right?

Thank you. Bye,
Vito

radek8, 2021-03-24:
Yes, your reasoning is correct.
If the current of one motor is 580mA and you connect 2 motors in parallel to one driver you have to double the current.
Attention. the engines must be the same. Otherwise there is a risk that one will be overloaded.
A better solution would be to use 2 TMC2209 drivers for the Z axis and connect each motor to a separate driver

Rincewind08, 2021-03-29:
unless you are DESPERATE, ignore those dual z ports. they might be good for color mixing, if you reassigned the extruder. I can't believe they keep putting them on there.

vitucciog, 2021-03-29:
Yes, I could also use the driver for the second extruder for the second z axis stepper, but using dual z is like the original configuration.

I'm currently testing it without problems with 900mA (800mA were not enough). Considering that the driver is rated up to 1,7A and that the z movements are few compared to x and y, I don't think it will overload neither steppers or drivers.

radek8, 2021-03-29:
How will you deal with the synchronization of both motors when they are unlocked?
Toothed belt?

radek8, 2021-03-29:
I have Ender3. I added a double extruder, but due to the greater weight, the steps were lost.
You replaced the original 42x34 engine with a new 42x40 engine.
He increased the current. He replaced the trapezoidal screw. The original pitch was 4mm. new screw 2mm. The problem with losing steps has been resolved.
For more precise movement, I added a second screw to the other end of the arm, connected by a toothed belt and it's great.
I think one more powerful engine is better than two weaker ones ....

vitucciog, 2021-03-29:
Originally the CR-10S has no synchronization. That's a defect for many people even if I did not suffer particular problems for that. Anyway I'm planning to add a belt with toothed pulleys on top of the z axis.