bigtreetech/BIGTREETECH-SKR-V1.3 #516: Blank LCD screen.

BoomerT6, 2021-03-27:
I have just upgraded my printer from an Arduino/Ramps 1.4 controller to a BTT SKR 1.4 Turbo. I am using the same 2004 LCD screen, so have left those settings unchanged in the Marlin configuration files. However, on first power up, when I expect the firmware to load, the screen remains completely blank. 
The board is live. LEDs are lit, the driver cooling fan runs and the endstops light when triggered. The Raspberry Pi I use with Octoprint struggles to initialise its connection with the printer and tends to loose it after a few minutes. 
I think the firmware flashed, as I was, via Octoprint, able to jog the X, Y and Z axis. I was also able to start a preheat and comunicate through Terminal, before the Pi lost the connection. I've tested the LCD with the old board and it works fine.
Any suggestions?

radek8, 2021-03-27:
What display driver do you use?
Submit configuration files to Marlin.

BoomerT6, 2021-03-28:
I'm sorry, but I do not know what you mean by display driver!?
I have visited the Marlin webpages, but cannot see how I might submit the configuration files.
I am on something of a learning curve here!

radek8, 2021-03-28:
Do you compile your own FW or download a precompiled bin file from the Internet?

Send
Configuration.h
Configuration_adv.h

radek8, 2021-03-28:
In the Configuration.h file, you must select the display driver according to the LCD you are using!

BoomerT6, 2021-03-28:
"#define REPRAP_DISCOUNT_SMART_CONTROLLER" - The same as I used with my Arduino/Ramps 1.4 controller.

There is no precompiled files to suit my printer, as it is one I have designed and built myself. 

[Configuration_adv.h.pdf](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/6218246/Configuration_adv.h.pdf)
[configuration.h.pdf](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/6218247/configuration.h.pdf)

radek8, 2021-03-29:
Make sure you have the EXP1 and EXP2 connectors oriented correctly

radek8, 2021-03-29:
Wouldn't you rather replace the display with a touch screen? :-)

BoomerT6, 2021-03-29:
I did that at the time of installation, as I noticed that left and right they are the opposite to the RAMPS.

radek8, 2021-03-29:
Yes, but they are also rotated 180 °

BoomerT6, 2021-03-29:
Not on mine. They will only go in one way.
Pluged into the RAMPS the screen works fine. EXP1 is on the right. Pluged into the SKR 1.4 the screen is blank. EXP1 is on the left.

radek8, 2021-03-29:
Connection is possible in 2 ways!
Some manufacturers place the cutout on the connector upside down.

radek8, 2021-03-29:
You need to find and compare the crater puin is 5V and GND

BoomerT6, 2021-03-29:
Looking at the boards from above with the connectors at the bottom.
On the SKR 1.4 the pins at the right of the connectors are identified as +5v and GND. On the RAMPS the pins to the right are identified as Pin 1 and Pin 2. It is my understanding that Pin 1 is +5v and Pin 2 is GND. In all cases the connector cutout is at the bottom. I think both follow the same convention.

BoomerT6, 2021-03-30:
I have only just noticed this =o) 
 
"Wouldn't you rather replace the display with a touch screen? :-)"

Yes I would, and I have. I use a 7" Raspberry Pi touch screen to control my printers. But, the other issue I have with this board is that the Pi has difficulty initializing and maintaining contact with it. When it does manage to connect, it drops the connection after just a few minutes. I do not have the same problem with my old Ender 3 Pro.

radek8, 2021-03-30:
https://reprap.org/forum/read.php?415,879300,879300

radek8, 2021-03-30:
Some displays may not be compatible :-(
What is your 2004 display?

https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/4

BoomerT6, 2021-03-30:
It does seem I am not alone with this problem. The Anet display is a different animal which I think has to be configured for in the pin file for the SKR 1.4.
My 2004LCD came with the RAMPS 1.4 and should be good with the defaults in the SKR 1.4 pin file. If pins 1 and 2 on EXP1 are good, then the display should at least light up me thinks, even if there is no data displayed!? It does not. I shall power up the board and see if there is power to the +5v pin or not. It could be as simple as that. Watch this space! =o)

radek8, 2021-03-30:
If it doesn't even light up, your connectors have been rotated 180 ° or are interchanged

radek8, 2021-03-30:
take a picture of the connection

BoomerT6, 2021-03-30:
Well there is +5v to pin. I think I will take the 2004LCD off the machine an see if I can determine just how it is wired up. I shall also check which pins provide power on the RAMPS connector board. If it is not pin 1 for power and 2 for ground, then that kinda goes against the norm.

BoomerT6, 2021-03-31:
The results were not as I expected. On the RAMPS pins 2 and 10 showed 5v. I have one ribbon lead that has a connector at 180 degrees. I tried it, and sure enough the LCD lit, thought it only showed solid character blocks. I then removed the pin shrouds from the pins LCD so that I could reverse both connections there. It lit, but still nothing displayed. My RAMPS and 2004LCD come from the same source and are clearly configured differently to the 2004LCD's that the SKR 1.4 Turbo support. Turning the connectors through 180 was not a fix, so I have no idea what the correct pin configuration is that I could change either physically, or in firmware. 
After all the head scratching and time wasted, I have decided that your suggestion of a touch screen is the best solution. I have decided to purchace a BTT TFT. That will be compatible with my BTT board I hope!
Thanks for your input radek8

radek8, 2021-03-31:
At least you can enjoy the new display.
The 2004 display is long overdue.
Then I recommend buying TFT35 V3.0, TFT E3 V3.0 or larger.
Be sure not to buy smaller or 3.5 "in versions 1 and 2
The displays also support the graphical simulated Marlin 128x64 mode.
You will no longer need an octoprint :-)

radek8, 2021-03-31:
I have SKRV1,4Turbo + TFT35 E3 V3,0
They work great

BoomerT6, 2021-03-31:
Ah, I have already ordered a TFT24 =o/  The TFT35 is the nicer display for sure, but more than I want to spend. I also want to stick with Octoprint, especially now that Autodesk have bought out a plug-in that allows you to slice and print directly from Fusion 360 via Octoprint.

radek8, 2021-03-31:
:-(
So good luck with TFT24

BoomerT6, 2021-04-10:
My particular 2004LCD was not compatible with the BTT SKR 1.4. I procurred a BTT TFT24 and that works fine.

closed 2021-04-10 by BoomerT6