bigtreetech/BIGTREETECH-SKR-V1.3 #517: Hotend Thermister reading constant temperature

dchristie1a, 2021-03-31:
The BTT SKR v1.4 Turbo board running Marlin 2.0 has been working great for about 2 months.  Now, I suddenly get a E1 Heater error for the hotend.  On further investigation, the temperature reading on power on of the board was 34 degrees.  When I power up the hotend, it does heat up, but the temperature remains at 34 degrees and so Marlin kicks in and errors out (as it should.

If I disconnect the thermistor the temperature still reads 34 degrees (instead of going to -14 degrees).  If I cycle the power, the temperature still reads 34 degrees without the thermistor connected.  Seems like there has been an internal HW failure in the board.

radek8, 2021-04-01:
According to your description, it looks like a damaged pin of the processor which measures the temperature of the hotend. These inputs are sensitive to voltages higher than 3.3V
If you use only 2 temperature sensors (1 hotend + 1 bed) you can use the third unused input for the temperature sensor.
All you have to do is change the port number for your board in the pin definition file.

If you need 3 temperature sensors, it is also solvable, but you will need a small HW adjustment, including soldering.

oldman4U, 2021-04-05:
Somehow it would be friendly to send radek8 who tried to help you an answer. Wouldn't it be!?