bigtreetech/BIGTREETECH-SKR-V1.3 #518: Ventilation for Cooling of the filament at 100%.

Magiquall, 2021-04-05:
I just received my SKR 1.4 Turbo and I just implanted it in my Hypercube which worked before with an MKS SGEN L.

I have several concerns, but one worries me. The cooling ventilation of the filament is at full blast all the time. I can't cut it from the screen let alone vary it.

As I am still working on the Marlin firmware, would you have to check a parameter before saying that the card has a fault.
A marlin parameter or a other try for see if the output of this FAN0 is not damaged.

Thank you

radek8, 2021-04-05:
Good day
which connector did you use for the cooling fan?

Magiquall, 2021-04-05:
Hello.

I connect the cooling fan to FAN0

radek8, 2021-04-06:
The FAN0 connector is the correct connector.
If the fan does not use the speed or shuts down, there will be a problem with the SKR board.
The switching transistor or pin P2.03 of the processor may be defective.
If you dare to solder SMS components, you can measure with a voltmeter where the problem is. if there is a problem with the transistor, you can replace it.

However, if you do not use the output on the HE1 hotend, you can remap the pin definition and use the HE1 to switch the cooling transistor.

If HE1 is not free, you can buy an external mofset and use one of the free pins that the PWM handles to control the cooling fan.

There are several options to correct the problem, it depends on you which one you choose :-)

Or you can claim the board to the supplier.

Magiquall, 2021-04-06:
Hello, 
The card is new, so I'm making a return, thanks for the information. 

I tried putting the fan on FAN1, for the e1 hotend, but it was running full too, but I didn't redirect the pins, so it might be normal.

closed 2021-04-06 by Magiquall

radek8, 2021-04-06:
Attention. The FAN1 connector is still connected to the power supply. Therefore, it cannot be regulated or switched off

radek8, 2021-04-06:
I still ask, did you define the SKR V1.4 board in Marlin correctly?

#define MOTHERBOARD BOARD_BTT_SKR_V1_4_TURBO

Magiquall, 2021-04-06:
Yes of course😉

Magiquall, 2021-04-06:
> Attention. The FAN1 connector is still connected to the power supply. Therefore, it cannot be regulated or switched off

This is what I understood too, my English is very limited, but it seemed to me that only the FAN0 was with PWM.