bigtreetech/BIGTREETECH-SKR-V1.3 #520: Board will not work with VDD but USB

oalkan2006, 2021-04-11:
I have a SKR 1.4 Turbo on CR 10 V2 and it was working fine until I connect tge board to a raspberry pi 3 b plus with octoprint. It did not make a connection with printer on web interface. Then I realised no red led, no lcd screen and nothing was working. I unplugged all the cables but the 24 vdc and the result is the same. I set the power to usb and connected it to computer. The board is working fine. I can also load firmware with SD card. Problem is something with the 24 v DC inut, i think. Any hopes to make that board work again. Thanks

radek8, 2021-04-12:
It seems that the octoprint overloaded the 5V power supply on the SKR board.
Check all fuses on the SKR board, especially on 5V branches.
If the fuses are OK and the 5V does not work, you have a problem with the 5V supply.
You try to repair the 5V power supply.
If you can't fix it, use a 5V external module for SKR V1.4
It will be necessary to break through some pins.

[image]

oalkan2006, 2021-04-13:
Thank you for your help. I checked the fuses with multi meter and it seems fuse 2 has blown. I just shorted those pins for now and I will be replacing it with 10 A fuse soon. Thank you!

closed 2021-04-13 by oalkan2006