bigtreetech/BIGTREETECH-SKR-V1.3 #521: How to debug firmware thru Visual Studio Code over the USB interface?

mink007, 2021-04-18:
Hi, 
I am looking for a tutorial on how to debug firmware by putting breakpoints in Visual Studio Code and USB interface (connected with the BTT SKR 1.4 Turbo). 

How to setup? What all packages are needed? Do i need configuration file?

Thank you in advance for helping.

radek8, 2021-04-19:
https://www.youtube.com/watch?v=Ze5h9TJjACw

mink007, 2021-04-19:
This video is creating the firmware.bin file and uploads to the BTT SKR 1.4 board thru SD Card.
What I am looking for is I installed the firmware.bin on the BTT board then I want to run the PIO Debugger or any debugger which will allow me to continue the marline code step by step, put breakpoints in the marlin code, step into the function, etc.

mink007, 2021-04-19:
I tried to build with PIO debug and get this error message. The build is successful but it  fails to connect thru USB. I think i need STLINK.

========================= 1 succeeded in 00:01:50.456 =========================
undefinedTraceback (most recent call last):
  File "C:\Users\username\.platformio\packages\tool-pyocd@0.801.0\pyocd-gdbserver.py", line 24, in <module>
    from pyOCD.tools.gdb_server import main
  File "C:\Users\username\.platformio\packages\tool-pyocd@0.801.0\site-packages\pyOCD\__init__.py", line 18, in <module>
    import board
ModuleNotFoundError: No module named 'board'
C:\Users\username\.platformio\packages\toolchain-gccarmnoneeabi\bin\arm-none-eabi-gdb.exe: warning: Couldn't determine a path for the index cache directory.
Reading symbols from D:\Marline_Projects\SKR1_4_TURBO\BTT SKR V1.4\Firmware\Marlin-2.0.x-SKR-V1.4-Turbo\.pio\build\LPC1769\firmware.elf...
PlatformIO Unified Debugger -> http://bit.ly/pio-debug
PlatformIO: debug_tool = cmsis-dap
PlatformIO: Initializing remote target...
.pioinit:13: Error in sourced command file:
:3333: No connection could be made because the target machine actively refused it.

radek8, 2021-04-21:
I don't think the current pootloader can do what you're talking about.

mink007, 2021-04-24:
In platformio.ini, I added below lines under [common_LPC] section:

**********************************
debug_tool = cmsis-dap
;debug_tool = custom
debug_server =
    openocd
    -f
    ../scripts/interface/stlink.cfg
    -f
    ../scripts/target/lpc17xx.cfg
**********************************

Then build in debug mode, it will generate firmware.bin file. Load that bin file onto the SKR 1.4T. Then run "debug"...

I am stuck at point where execution does not halt at breakpoint ..

mink007, 2021-04-25:
https://community.platformio.org/t/need-help-getting-stlink-talking-to-programming-an-skr1-4t-3d-printer-cpu/14118/34?u=mink007

radek8, 2021-04-25:
I'm always learning something new :-)
Sorry I can't help you with this.

mink007, 2021-04-25:
STLink V2 is needed to put SKR 1.4 Turbo in debug mode. More detail on setup in PlatformIO at this thread.
https://community.platformio.org/t/need-help-getting-stlink-talking-to-programming-an-skr1-4t-3d-printer-cpu/14118/34?u=mink007

closed 2021-04-25 by mink007