bigtreetech/BIGTREETECH-SKR-V1.3 #526: Hotend turns on as soon as I power my printer

baldaaaa, 2021-05-07:
Hello everyone,

So I was wiring my DIY CR-10 printer and everything was ok. I tested everything with success for a couple of days but my sk4 v1.4 turbo board suddenly burned and now the hotend starts to heat up uncontrollably as soon as I power the printer.

**The component in the following picture is what burned:**
[image]

I don't know exactly what this component is. Just by looking it seems to be connected to FAN0 but I might be mistaken. Maybe someone can help me pointing out what this does and what could be the reason for this happening.

Obviously the board is damaged now but I have another one and I don't want to mess that up too.

radek8, 2021-05-07:
It is strange
The transistor that switches the fan burned down. It has nothing to do with hotend heating.
I have a question
is the LED indicating the hotend on?
If not, the problem could only be in the burned out transistors that could be replaced or connected externally.

radek8, 2021-05-07:
Couldn't there be a short in the cooling fan?

radek8, 2021-05-07:
[image]

baldaaaa, 2021-05-07:
Thanks so much for your response @radek8 that is exactly the component I was watching in the schematic but have no idea what might have caused it. The LED is reporting temperatures normally. My only guess is I shorted it because I didn't mount the board anywhere but was testing it over my desk on the bare wood and there might have been humidity or something.

From what I understood (correct me if I'm wrong) the MOSFET is dead and current is passing through and entering MC74HCT125ADTR2G through pin 11?

How would this be fixable? I guess just change the MOSFET? Where do I get such a component?

Thanks for your help :)

radek8, 2021-05-07:
The A01T fan mosfet is definitely defective

I assume that the MC74HCT125ADTR2G circuit will also be defective, because it switches on the hotend heating when it does not heat.
But I'm not sure for sure. you would have to measure the input and output voltages in the open and closed states.

https://www.aliexpress.com/item/32982807880.html?spm=a2g0o.productlist.0.0.1f7d3d73ZPUmLt&algo_pvid=97cff357-1116-402f-88fe-2161b4a5fcaf&algo_expid=97cff357-1116-2161616162 searchweb0_0, searchweb201602_, searchweb201603_

https://www.ebay.com/itm/332327703984?hash=item4d604655b0:g:KmEAAOSwzRlZgGeq

baldaaaa, 2021-05-07:
Thanks a lot for your help. I'll go ahead and close this issue.

closed 2021-05-07 by baldaaaa

radek8, 2021-05-07:
[image]