bigtreetech/BIGTREETECH-SKR-V1.3 #527: Skr 1.4 turbo firmware updating

Shayner2005, 2021-05-08:
SKR 1.4 turbo will change the .bin file to .cur file fine, but will not apply changes to actual board firmware. Tried both through SD card and with octoprint's firmware updater plugin.

radek8, 2021-05-08:
Initialize eeprom

Shayner2005, 2021-05-08:
> Initialize eeprom

Can you elaborate?

Cheo7811, 2021-05-08:
@Shayner2005 you don't mention what printer and display you are using but if the option to Initialize EEPROM is not available on your display you can connect via USB to your car and use a serial program like Pronterface, once connected to your printer issue M502 which will reset the EEPROM to the values in your firmware.

Shayner2005, 2021-05-08:
> @Shayner2005 you don't mention what printer and display you are using but if the option to Initialize EEPROM is not available on your display you can connect via USB to your car and use a serial [...]

Ender 3 with a btt tft35 v3. Having screen issues prompted the firmware update. I use octoprint, however.

Cheo7811, 2021-05-08:
Then you can send M502, M500 from Octoprint Terminal and see if that fix it.

Shayner2005, 2021-05-08:
> 
> 
> Then you can send M502, M500 from Octoprint Terminal and see if that fix it.

Nope, no change.

Dirge6, 2021-05-17:
This just now happened to me. Prusa TallBear w/ Phaetus Dragon, BTT 1.4Turbo, TMC2209s V1.2, and TFT43. Resetting EEPROM didn't work.  M504 shows OK.

dirgesan, 2021-05-18:
An update: I believe my problem is solved. Seems to be updating. I rebuilt firmware with latest Marlin Bugfix. This also gave me a chance to finally install an AT24C with a 512 chip on the motherboard.

radek8, 2021-06-04:
please close this ticket

closed 2021-06-05 by Shayner2005