bigtreetech/BIGTREETECH-SKR-V1.3 #530: Marlin firmware not showing board

jayala1213, 2021-06-02:
I'm using an Skr 1.4 turbo board with Marlin firmware but boards.h isn't showing my board an an option. The closest I see is a skr 1.3. Is the skr 1.4 turbo no longer compatible with marlin?

radek8, 2021-06-03:
I see him there :-)
Attention:
SKR V1.3 has an LPC1768 processor
SKRV1,4 Turbo has an LPC1769 processor

Cheo7811, 2021-06-03:
Yes, maybe it is better this way:
[image]

jayala1213, 2021-06-03:
Hmmm that's really strange, mine doesn't show it. The closest it shows is skr 1.3. [image]

Cheo7811, 2021-06-03:
That list looks very old, are you sure you have downloaded the latest Marlin 2.0.8.2 by pushing the CODE button and them the Download.ZIP?
[image]

jayala1213, 2021-06-03:
> That list looks very old, are you sure you have downloaded the latest Marlin 2.0.8.2 by pushing the CODE button and them the Download.ZIP?
> 
> [image]
> 
> 

Yeah I downloaded that version. Do you think that's the problem? I could've downloaded the wrong one.

radek8, 2021-06-03:
@jayala1213 
You have version 2.0.0

radek8, 2021-06-03:
https://marlinfw.org/meta/download/

jayala1213, 2021-06-03:
> https://marlinfw.org/meta/download/

Ok thank you so much! I'll try it!

closed 2021-06-04 by jayala1213