bigtreetech/BIGTREETECH-SKR-V1.3 #531: Z-axis won't move to max position - marlin 2.0.x

fabricioffo7, 2021-06-16:
hi I'm very new to this
I have this problem, the z-axis does not move to the maximum position I already tried with G-code commands from pronterface, tried from the touch screen terminal, and tried from marlin mode, each attempt the maximum that it moves is 310mm, and in marlin 2.0 in configuration.h I already put z_max_pos 410mm and z_min position 0
also, change the display firmware config.h (tft35 3.0) with the same values. I do not know what else it could be, look for all possible information and try with various things but nothing changes, it seems that it is blocked from the factory or something like that I do not understand, please I need help
[image]
![image](https://user-images.githubusercontent.com/589
[image]
90459/122243000-c1af4100-ce91-11eb-97ee-5d7696c0d3c2.png)

Cheo7811, 2021-06-16:
Hola Fabricio:
What printer do you have? is the printer printing properly? could be that the steps for the Z drive are wrong, maybe you can post the Configuration.h and Configuration.adv.h

Saludos,
Julian

fabricioffo7, 2021-06-16:
hi
I do not know if the problem is the steps / mm because I already measured and calibrated these and when I raise the z-axis there are about 100 mm of travel.
[Configuration_adv.md](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/6664289/Configuration_adv.md)
[Configuration.md](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/6664290/Configuration.md)

> Hola Fabricio:
> ¿Qué impresora tienes? ¿La impresora está imprimiendo correctamente? podría ser que los pasos para la unidad Z sean incorrectos, tal vez pueda publicar Configuration.hy Configuration.adv.h
> 
> Saludos,
> Julian

Cheo7811, 2021-06-16:
The files look ok, I however don't understand why you are changing things in the TFT firmware files, the configuration is done in the config.ini that you edit and place on the root of a SD card together with the firmware.bin file and the TFT folder for the theme you want to use, since you are no defining M115 you set the size in the config.ini file here:
[image]

fabricioffo7, 2021-06-17:
> The files look ok, I however don't understand why you are changing things in the TFT firmware files, the configuration is done in the config.ini that you edit and place on the root of a SD card [...]

thanks @Cheo7811 Cheo7811 I also modified the config.ini file and nothing works the printer moves well in the X and Y axis but in the z it does not move up to 410mm
[image]