bigtreetech/BIGTREETECH-SKR-V1.3 #532: #define E1_AUTO_FAN_PIN

dimaknight1109, 2021-06-22:
I would like to add my hotend fan to the following which can be found in configuration_adv so that the fan turns off when its lower than 55 degree and turns on which it reaches 55.

#define E1_AUTO_FAN_PIN

How do i make this work. what do i need to type after "#define E1_AUTO_FAN_PIN"

dimaknight1109, 2021-06-22:
#define E1_AUTO_FAN_PIN  P2_04 would this be correct?

radek8, 2021-06-23:
You didn't write what kind of board you had

hopeinformer, 2021-06-23:
#define E0_AUTO_FAN_PIN P2_04
That's the one that works for my SKR v1.4 & the Turbo, then you set the value in the line below
#define EXTRUDER_AUTO_FAN_TEMPERATURE 50, or in your case "55"
I suppose you could use the E1 for the fan instead and set your hotend thermistor to E0 but I found it easier the other way around.
But radek8 is right, this pin will depend on which board you're using. Just look up the schematics and pins for your board to find the right one.

dimaknight1109, 2021-06-23:
now I'm confused, lol
I read on some documentation i found online that i need to do the following

define E0_AUTO_FAN_PIN FAN1_PIN

Does not mention anything about adding a pin number.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

radek8, 2021-06-23:
You have to define the pin as @hopeinformer writes.
If you don't define a pin, Marlin can't start the fan because it doesn't know where it's connected

dimaknight1109, 2021-06-23:
I get that but on the documents that can with bigtreetech they does state
you need to add a pin number. Reason why I'm confused.

radek8, 2021-06-23:
Yes, the SKRV1.3 board does not have a dedicated connector or MOFSET for the extruder fan.
The connector for switching the second hotend is unused (unless you use two hotends) and can be redefined to connect the fan  extruder cooler.
What's incomprehensible about that?

dimaknight1109, 2021-06-23:
This is what the document states to use the hotend fan
define E0_AUTO_FAN_PIN FAN1_PIN

This is what I pickup on when watching a YouTube video
#define E0_AUTO_FAN_PIN P2_04

This is my confusion. Which one is right?

radek8, 2021-06-23:
Thirdly, I say that this is correct
#define E0_AUTO_FAN_PIN P2_04

You must replace "FAN1_PIN" with the pin number to which you connect the fan.

Connect the fan to the connector as shown above

radek8, 2021-06-23:
You can also turn off and control the fan speed, which cools the control board with the motor drivers, but you would have to buy an external Mofset to switch the fan.

radek8, 2021-06-23:
It would not be the first mistake in Bigtreetech's documentation.
Bigtreetech states that the fan is connected to the pins of the FAN1 connector
But this connector has a continuous power supply.

[image]

dimaknight1109, 2021-06-23:
So it's #define E0_AUTO_FAN_PIN P2_04

dimaknight1109, 2021-06-24:
apologies have one more question,
For dual Z, im using the E1 motor. I have defined Dual Z in Configuration
_adv. but for defining the stepper motors. do i add TMC2208:

#define E1_DRIVER_TYPE TMC2208_STANDALONE
or
#define Z2_DRIVER_TYPE TMC2208_STANDALONE

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

radek8, 2021-06-24:
What is your reason for using STANDALONE mode?
SKR boards are prepared for UART mode which can be configured at runtime

dimaknight1109, 2021-06-24:
I have seen videos on this but dont know much about what to do for each
modes. Standalone seem to be the easiest.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

radek8, 2021-06-24:
Watch instructional videos about UART mode :-)

dimaknight1109, 2021-06-24:
sure ill check som out, but what the difference with UART and standalone

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

radek8, 2021-06-24:
Monitor Trinamic drivers
   * for error conditions like overtemperature and short to ground.
   * To manage over-temp Marlin can decrease the driver current until the error condition clears.
   * Other detected conditions can be used to stop the current print.
   * Relevant G-codes:
   * M906 - Set or get motor current in milliamps using axis codes X, Y, Z, E. Report values if no axis codes given.
   * M911 - Report stepper driver overtemperature pre-warn condition.
   * M912 - Clear stepper driver overtemperature pre-warn condition flag.
   * M122 - Report driver parameters (Requires TMC_DEBUG)
   * M913 - TMC Hybrid Threshold Speed
   * M569 - TMC StealthChop