bigtreetech/BIGTREETECH-SKR-V1.3 #533: Skr1.3 CR20 display/uSD card problem

creatron, 2021-06-22:
The CR20 Pro comes with a BLTouch and 12864 mini clone. I use Marlin 2 bugfix and can either get the screen to work, then no uSD card access, or uSD card but then no screen. 
Common setting
#define SDSUPPORT
For the screen to work you need to set 
//---------------------------
#define MKS_MINI_12864
#define LCD_CONTRAST_MIN  160 
#define LCD_CONTRAST_INIT 180
#define LCD_CONTRAST_MAX  240 
and 
 #define SDCARD_CONNECTION LCD // LCD
Now the screen works but no SD card access
//------------------------------
If however, you change 
#define SDCARD_CONNECTION ONBOARD 
The screen does not work, but you have access to the uSD card.
Is it possible to have both available as upgrades for Creality CR20 pro.  I have silent steppers, with BLTouch, Neopixels, etc working but need a display and SD .. If I use a stock Creality from my Ender/CR10 everything works, as well as 
SBase 12864,, but the Creality mini 12864 is problematic.

IndiamTek, 2021-09-17:
Hi Creatron,
the problem is still with my Ender3 stock LCD and SD card. I am using SKR 1.4 Turbo here.
Can you tell me hows your CR10 LCD and sd card are working together?  
[image]

creatron, 2021-09-17:
I submitted a bug on Marlin, but it is not yet accepted, they did change
some code but it still did not work. I can however send you my solution to
try with no guarantees.

-- 
Gerrie de Jager

(Quite people have the loudest minds.
Whether you think you can or you think you can't -- your'ed right.
Don't confuse having less with being less, having more with being more.
Who you are is far more important than what you have.
Ask busy people, others are 'busy')

Cell  072 198 1678

*www.creatron.co.za <http://www.creatron.co.za>*

This e-mail, its contents and any file attachments transmitted with it are
intended solely for the addressee(s) and may contain confidential
proprietary information. Access by any other party without the express
written permission of the sender is unauthorised. If you have received this
e-mail in error you may not copy, distribute or use the contents,
attachments or information in any way. Please destroy it and contact the
sender.

IndiamTek, 2021-09-17:
please send the solution. i will try.