bigtreetech/BIGTREETECH-SKR-V1.3 #534: SKR 1.4 Turbo: The E0 heater output does not work

Vertorek, 2021-06-25:
Hello, 

After two days of not using a printer, the E0 heater stop working is not heating up. Pins are not supplied with 24V, diode indicating working cycles is off whole the time.

The fuses are ok, heating the bed is fine too but I can not heat up the nozzle. 

I thought that is a firmware issue I made some minor changes but not related to pins. I had a second board (2) I flash the firmware and connected the thermistor, 24V power supply and heater. All was fine heater E0 was working. I loaded the same firmware to the first board again no changes E0 was dead. 

When I tried with the second board again E0 was dead too.

So I have two boards with the same issue

I notice that people complain about such issues but I did not find anything useful.

radek8, 2021-06-25:
Can you measure the electrical voltage at Gate transistor Q3?
in the house when the heating nozzle is switched off and when the nozzle is to heat

[image]

radek8, 2021-07-27:
If you no longer have a problem, can you close the ticket?