bigtreetech/BIGTREETECH-SKR-V1.3 #535: extuder no responding

dimaknight1109, 2021-06-27:
I have a skr 1.3 board, 
I uploaded my firmware with the steps i followed from chris reliy's youtube video.
1) Everything else seems ok X home and moves well, same with Z
but when i move my bed, its moves smoothly but when i home y axis, it moves the bed a little and nothing, will not home

2) My extruder will not respond. I checked cables, motor, and the driver and they work fine. I wondering if there is something in marlin i need to define maybe.

I want to  mention I set my TMC2208 to 1.1v. I have no idea what is a good number

please i need help.

radek8, 2021-06-28:
Why do you have Endstop on the X axis defined in the opposite way to the Y axis?
#define X_MIN_ENDSTOP_INVERTING true  
#define Y_MIN_ENDSTOP_INVERTING false

If you use the same wired microswitch, it should be defined the same.

radek8, 2021-06-28:
TMC2208 drivers are in STANDALONE mode
I personally prefer UART mod

radek8, 2021-06-28:
Which means point 2 that you can't move the extruder. Does this mean that the fiber is not extruded during printing?

radek8, 2021-07-27:
If you no longer have a problem, can you close the ticket?

closed 2021-07-27 by dimaknight1109