bigtreetech/BIGTREETECH-SKR-V1.3 #537: SKR 1.3 working on USB but not working on AC.

StefanoV1989, 2021-07-12:
I updated my CR10 Mini from the standard board to a Bigtreetech SKR 1.3 and i installed TMC2208 in UART with a 5th extruder motor (and a mixing hotend ZSYong Mix),

I plug everything correctly and double checked. I uploaded the firmware.bin (became FIRMWARE.cur in the SD) and tried connecting with pronterface via USB cable with switch power on USB. 
Pronterface says "printer is online".

I then remove the USB cable, place the switch on AC Power and switch on the printer. Fan turned on and everything seems OK, but:  display is off and if i press the display button i get a "Beep",

What can i do? Is there a debug for SKR 1.3 to check if there is any conflict?

radek8, 2021-07-12:
Do you have the switch in the correct position?
Did you check the fuses?

[image]

StefanoV1989, 2021-07-13:
> 
> 
> Do you have the switch in the correct position?
> Did you check the fuses?
> 
> [image]

Yes, switch OK and Fuse working. 

My Cr10-Mini has the double cable monitor as for CR10S (EXP 1 and EXP 2 plugs). I read online that i need to 180 degree rotate both cables on the SKR 1.3, is that true?

StefanoV1989, 2021-07-13:
I rotated the EXP1 and EXP2 cable, now the display is on but blank, and the board started to do a long continuous beep and i can't recognize it anymore as COM port on my PC...
I don't know what to do now...

radek8, 2021-07-13:
If the display worked, why did you rotate the connectors?
Return the connectors to their original position. There is a risk of damage to the electronics.
You write that the switch is OK. That's a funny answer.
In what position is it?
The correct position is this:

[image]

radek8, 2021-07-13:
What display model do you use?
What display driver did you define in Marlin?

StefanoV1989, 2021-07-14:
> 
> 
> What display model do you use?
> What display driver did you define in Marlin?

Display model is the CR10s display, in Marlin i tried REPRAP_DISCOUNT_FULL_GRAPHIC_LCD but also CR10_STOCKDISPLAY.
Both didn't work.

radek8, 2021-07-14:
Send a photo of the connector connection

StefanoV1989, 2021-07-14:
> 
> 
> Send a photo of the connector connection

[image]

radek8, 2021-07-14:
use
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

radek8, 2021-07-14:
You're right with the cables. In the photo, the cables are rotated 180 °
Connect them correctly.

radek8, 2021-07-14:
[image]
[image]

StefanoV1989, 2021-07-14:
@radek8 I tried rotating 180° but the display switches on with really low contrast for 2 seconds and then off. If i press the encoder, it beeps and nothing else.
I tried to re-connect the display on the cr10mini board and it works properly, so i wonder if it's a board problem.
I ordered another display from Amazon (Reprap full graphic lcd), i will try with it when i will get it this Friday. If it doesn't work either, then i will know it's a board problem.

radek8, 2021-07-15:
Did you use the correct driver?

#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

I bought a touch screen BTT TFT 35 E3 V3.0
I practically don't use Marlin mode anymore.

StefanoV1989, 2021-07-15:
> 
> 
> Did you use the correct driver?
> 
> #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
> 
> I bought a touch screen BTT TFT 35 E3 V3.0
> I practically don't use Marlin mode anymore.

Sure i did.
Tomorrow i will receive the generic Reprap full graphic display from Amazon. If it doesn't work either, i decide i will upgrade everything to a SKR 1.4 Turbo + TFT35 v3.0 and that's it.
I payed SKR 1.3 just 17$ on Aliexpress from a random seller (not BTT), i wonder if there's some problem on it ... better if i buy everything from BTT page next time...

radek8, 2021-07-15:
If you turn on SKR V1.3
without a display and you connect to a computer via USB, is your COM port mapped?

radek8, 2021-07-15:
> Sure i did.
> Tomorrow i will receive the generic Reprap full graphic display from Amazon. If it doesn't work either, i decide i will upgrade everything to a SKR 1.4 Turbo + TFT35 v3.0 and that's it. [...]

TFT35 E3 V3.0 is better for you. It is dimensionally compatible with the original Creality display.

StefanoV1989, 2021-07-15:
> 
> 
> If you turn on SKR V1.3
> without a display and you connect to a computer via USB, is your COM port mapped?

Yes sure. and i can also move axis etc. i didn't try to print but i think i can normally print because everything seems OK.

radek8, 2021-07-15:
So Marlin is running.

StefanoV1989, 2021-07-16:
Ok, everything is working now with the new screen from Amazon.
I also got a problem with mizing extruder but i discover it was a Marlin 2.0.9 bug, i downgraded to Marlin 2.0.8.2 and everything is working now!

radek8, 2021-07-17:
If you have resolved the issue, you can close the ticket.

closed 2021-07-17 by StefanoV1989