bigtreetech/BIGTREETECH-SKR-V1.3 #538: [SKR 1.4 Turbo] No serial connection detected via USB. USB power works

Garandaiz, 2021-07-13:
Hi everyone,

Until recently, the board worked perfectly. But lately, when plugged via USB to a computer, no serial port is recognized. I tried to change the jumpers to the USB powered position and the board is in fact powered, but still the computer doesn't detect any serial port.

Thank you very much in advance.

radek8, 2021-07-13:
Do you have a port defined in Marlin?

Garandaiz, 2021-07-14:
I have the following definitions in Configuration.h:
`#define SERIAL_PORT -1

#define SERIAL_PORT_2 0`

radek8, 2021-07-14:
Either you have a damaged USB port on the SKR board, or you have problems with the drivers on your computer.
Make sure that "unknown device" does not appear on your computer when you connect the SKR to your computer

radek8, 2021-07-15:
That looks fine