bigtreetech/BIGTREETECH-SKR-V1.3 #542: skr v1.3 - no USB connection with PSU power

jongbot, 2021-08-13:
I cannot connect to the SKR 1.3 via USB unless the power jumper is on 5V + USB. I also realized the firmware.bin can only be loaded when the jumper is on 5V + USB. If I put the jumper back to INT + 5V, which is instructed for running the printer.. I can't connect to USB.

philocritus, 2022-06-27:
> I cannot connect to the SKR 1.3 via USB unless the power jumper is on 5V + USB. I also realized the firmware.bin can only be loaded when the jumper is on 5V + USB. If I put the jumper back to INT + 5V, which is instructed for running the printer.. I can't connect to USB.

I'm having the same problem. 
Did you manage to solve it or was it bricked? 
thank you!

jongbot, 2022-06-29:
> I'm having the same problem. Did you manage to solve it or was it bricked? thank you!

I found out later my psu was failing to provide full power. It was only enough to turn on some LEDs but would fail everything else. After I swapped out the psu, it all worked.

philocritus, 2022-06-29:
> I found out later my psu was failing to provide full power. It was only enough to turn on some LEDs but would fail everything else. After I swapped out the psu, it all worked.

Thank you for the reply! I ended up finding that I had a blown fuse!
its sorted now!

Genti-m, 2022-07-09:
I want to unsubscribe please.

Thank you