bigtreetech/BIGTREETECH-SKR-V1.3 #547: 1.4 turbo - blank screen on firmware compile

sdrulik, 2021-09-14:
Hello,
I've ran into a peculiar issue where I get a blank screen on compile for my Ender 3. I used the guide to set up TMC 2209 drivers, sensorless homing, BLTouch, Z probe homing, and using the servos/probe for the BLTouch. Using Marlin 2.09 and the latest Ender 3 1.4 configuration from marlin.
[https://media.githubusercontent.com/media/GadgetAngel/SKR-V1.4-Turbo-Stepper-Driver-Jumper-Configuration-Manual/master/CURRENT-Manual/SKR%20V1.4%20TURBO%20Stepper%20Driver%20Jumper%20Configuration_version%20V4.0_reducedFileSize.pdf](url)

Here are my config.h and config_adv.h files. 
  
[Configuration - Copy.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/7158328/Configuration.-.Copy.txt)

[Configuration_adv - Copy.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/7158331/Configuration_adv.-.Copy.txt)

I'm not sure what is wrong and if maybe my BLtouch wiring is wrong? I was given a 5 pin connector and not 3+2 pin connectors. My wires also are different colors (blue, red, yellow, white, black). I wired them by removing the 5 pin and converting it to 3+2 connector in the following wire order:

BLTOUCH wiring left -right (white, black, yellow. red. blue)
Servos port wiring  left-right (white, black, yellow)
probe port wiring left-right (red, blue)

Thanks!

radek8, 2021-09-14:
For Chinese 3Dtouch clones, it is not the color of the wire that matters, but the order of the contacts in the connector

https://github.com/KevinOConnor/klipper/issues/3574

picknell, 2021-11-07:
i basically did the same setup. it happened few times. that a firmware i did went fine, compiled a new one with minor change, without change to cabling and with the new version it got stuck on the blank screen.
[Configuration.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/7492200/Configuration.txt)
[Configuration_adv.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/7492201/Configuration_adv.txt)
here are my files for the last working version if it helps. got it working with trianglelabs 3d touch using the dedicated  bltouch slot. I think after all I solved it i think with formating the sd card, placing the bin file on it again and before taking the card out I used the safely remove feature on windows. Seems like that was the issue, not those minor changes between firmware.

Hetothet, 2024-01-27:
i am struggling with the exact same issue i manage to get a working screen and menu using the ender 3 pro profile but z axis homing did not work with the bl touch nor the endstop