bigtreetech/BIGTREETECH-SKR-V1.3 #550: Strange layershifts

Magnum8101, 2021-11-08:
I got some strange layershifts egen connecting SKR 1.4 T to my Anet A8. With original motherboard the prints get just fine.
I have tried 2 different turbo boards, 5 cheap TMC2209 steppers, and last try, BTTs TMC2208, but with similiar issues. Last try i did not tune the stepmotors. I am starting tonä give Up a silent printer with BTT SKR. 😔
[image]
[image]

AFactor2, 2021-11-08:
have you looked at the driver setup and micro stepping?I have several A8 units all on OE boards a BTT is a bit of a over kill for that machine id the r something you are looking to achieve? I have one from about 2013 and it has TL board's on X and Y, Dual PSU and MOS fet and i fixed the Z and the loos liner rods it prints ABS all day at 0.12 x 0.2 by 30mm SECSincerelyAndrey ***@***.***: This e-mail, including its contents and attachments, if any,are confidential.If you are not the named recipient please notify the sender and immediately delete it. You may not disseminate, distribute, or forward this e-mail message or disclose its contents to anyone else.   E-mail transmission cannot be guaranteed to be secure or error-free. The sender therefore does not accept liability for any errors or omissions in the contents of this message which arise as a result of e-mail transmission. We make no representation or warranty as to the absence of viruses in this e-mail or any attachments. For the protection of our clients and our business, we may monitor and read e-mails sent to and from our server(s).

#550)

Magnum8101, 2021-11-08:
Du you have a Good guide for driver setup and micro stepping? What i want to achive is furst of all a silent 3d printer, better Interface with touch screen etc.

radek8, 2021-11-09:
Send configuration.. Send a photo of the jempers set up under the stepper controllers

radek8, 2021-11-09:
What kind of step engines do you use?

Magnum8101, 2021-11-10:
I use original enginges from Anet.
[image]
 
I did renamed the files from .h to [.txt](url)

[Configuration.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/7515721/Configuration.txt)
[Configuration_adv.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/7515722/Configuration_adv.t
[image]
xt)

radek8, 2021-11-10:
Please take a picture of the TMC2209 driver from the bottom

Magnum8101, 2021-11-10:
I am now using TMC2208s now...
[image]

Magnum8101, 2021-11-10:
I am now using 2208s
[image]

radek8, 2021-11-10:
I see a problem here. The TMC 2208 drivers are not in UART mode.
You have to solder the two pins.

[image]

radek8, 2021-11-10:
[image]

radek8, 2021-11-10:
I would also set a higher current in the motour configuration, you currently have 580mA
Set 650mA. If it doesn't get too hot, you can go up to 800mA, but always check if it's not warming up,

radek8, 2021-11-10:
You only have HYBRID_THRESHOLD set to 45 mm / s.
I would set more, otherwise you won't enjoy the silent mode much :-)

radek8, 2021-11-11:
Some news?

Magnum8101, 2021-11-11:
Yeah! Now it is mutch better! Thank you very mutch! I connected these jumpers, with a waaaay to big soldering iron.
the 2208s are getting very hot, and so are my motors also. I turned down from 800mA to 700mA, maby I have to go lower.

radek8, 2021-11-11:
Set the current value so that it is only slightly warm after an hour of printing.
The 2209 drivers are better than the 2208 and heat up less.
If you have them too, use them. You will need to change the parameter in the configuration and can also modify the jumper on the 2209 drivers.

radek8, 2021-11-11:
What source do you use? according to the configuration do you have 12V power supply?

Magnum8101, 2021-11-11:
My motors should have 800mA on paper. but I will have to try lower Current now. My power supply is 12V.
The TMC2209 is cheap ones i ordered, someone called the "dummy" stepper. There are no jumpers i can sold on them.

radek8, 2021-11-19:
Send a photo to TMC2209

radek8, 2021-11-19:
Power supply 12V :-(

Magnum8101, 2021-11-22:
Thank you very much for your help! I am closing this thread.

closed 2021-11-22 by Magnum8101