bigtreetech/BIGTREETECH-SKR-V1.3 #551: BigTreeTech SKR 1.4 Turbo TFT35 TMC 2209 x 4 on a HEVO printer

bens888, 2021-11-12:
1. Using the bigtreetech side on TFT35, the heatbed and extruder will not heat up, and no movement on any axis.  A M503 command shows the setting and then showing repeated command of M155 S3 for pages and pages...
2. Using the Marlin side on TFT35, the heatbed and extruder will heat up, and there are movement on all axis including homing.  Tried to print from the SD card and the TFT will reboot... 
3. Attempted to reload the compiled firmware, but shutting down and system and restarting (power on), the board will not read the firmware.bin file.

Any help is appreciated!

radek8, 2021-11-13:
Please insert Marlin configuration file (configuration.h and configuration_adv.h) and TFT configuration (config.ini)
send photos of cable connections between SKR and TFT.
Thank you

bens888, 2021-11-14:
Here are the files.
![20211113_172934](https://user-images.githubusercontent.com/94185650/141664289-4adc9266-09b4-4818-b14f-3e791ca062c2.jpg
[image]

[ConfigFiles.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/7533037/ConfigFiles.zip)
t.com/94185650/141664290-2eb696f6-751d-4e03-8fc5-a3b54dc4a256.jpg)
)
..

bens888, 2021-11-14:
In the config.h and config_adv.h files if uncommented on 11Nov2021 did not successfully flashed to the SKR 1.4 Turbo PCBA.

radek8, 2021-11-14:
> V souborech config.h a config_adv.h, pokud nebyly 11. listopadu 2021 bez komentáře, nebyly úspěšně flashovány na SKR 1.4 Turbo PCBA.

I do not understand

radek8, 2021-11-14:
if you disconnect the two gray cables from connectors EXP1 and EXP2, will the display still be powered on?

radek8, 2021-11-14:
In Marin uncomment:
#define M115_GEOMETRY_REPORT
#define M114_DETAIL   
#define REPORT_FAN_CHANGE 
#define HOST_ACTION_COMMANDS
#define HOST_PROMPT_SUPPORT
#define HOST_START_MENU_ITEM
#define EMERGENCY_PARSER
#define SERIAL_FLOAT_PRECISION 4
#define LONG_FILENAME_HOST_SUPPORT
#define SCROLL_LONG_FILENAMES

radek8, 2021-11-14:
The TFT configuration is OK.
The cable connections also look fine.
Edit the port configuration in Marlin and uncomment the parameters I wrote.
Test whether the display connection works in touch mode.

bens888, 2021-11-15:
With EXP1 and EXP2 disconnected, the display still powers on.
I'll attempt to reload the compiled FW onto the PCBA. Previous attempts the board did not read the SD micro card to update the firmware.

radek8, 2021-11-15:
PCBA = SKR1. 4 Turbo?

radek8, 2021-11-15:
Reconnect cables EXP 1 and EXP2.
Do you have a working Marlin with the modification I sent?

bens888, 2021-11-15:
Yes, that's correct.
The PCBA was reset with the M502 command.
The PCBA was set to use USB on the jumper setting. PlatformIO did not recognize the port.  Changed the platformio.ini file for the "upload_port" to G: first and if failed to upload the firmware.  Changed the "upload_port" to com{113], com11 was the new port shown in device manager (windows 10).
With the USB cable connected to the laptop, the G: drive shows up and the latest firmware.bin file is present.
Attempt to load FW with the jumper on the PCBA moved back to "VDD" position, firmware.bin installed onto a Samsung SDmicro card, shutting power and turn-on power, the PCBA did not load the new firmware.

radek8, 2021-11-15:
I don't need to connect the SKR 1.4Turbo board to a PC to compile the new Marlin.
Just have FW compiled for LPC1769
copy the generated FW (Marlin \ .pio \ build \ LPC1769 \ firmware.bin) to the SD card (less than or equal to 8GB, FAT32 format)
Insert into SKR 1.4Turbo and reset.
The FW must be loaded, after loading the file is renamed to CUR.

bens888, 2021-11-15:
The original 128MB SDmicro card is not recognized by my laptop.  The Samsung SDmicro card is 32GB, so it's to large for the SKR board to recognize.  Will try to find a 8GB SDmicro.

radek8, 2021-11-15:
There may be a problem with large SD cards. A 32GB card may or may not work.
Smaller cards are more likely to be compatible

bens888, 2021-11-15:
8GB from Amzn will arrive tomorrow.

radek8, 2021-11-15:
Tomorrow? You probably have an Amazon warehouse in the garden :-)

radek8, 2021-11-15:
Marlin can't update. And did you manage to update TFT 35 to the latest versions?

radek8, 2021-11-15:
Prepare a new version of Marlin. The configuration you sent at the beginning of the conversation is out of date.

bens888, 2021-11-15:
TFT35 compiled and uploaded on Oct 04 2021.

bens888, 2021-11-18:
@radek8: Ahh the printer is working again with the new SDmicro card for the board to read and load the updated firmware.  Thank you sooooo much!!! 
Do I need to open another issue if the filament runout is not working... the M119 shows the switch is working, "triggered" for out of filament and "open" for filament present.

closed 2021-11-19 by bens888