bigtreetech/BIGTREETECH-SKR-V1.3 #552: SKR 1.4 + TFT24 + DCDC MODE Short circuit between power supplies

gabinet2708, 2021-11-21:
Hello, i was testing my new skr 1.4 board with the TFT24 accessory (conected with the Serial and Exp1 and Exp2), works fine, after that i tried the DCDC MODE Power Supply (with the TFT24 Disconected and the NPWR jumper moved to VOT) everything works fine.

When i tried the TFT24 connected with the Serial and Exp1 and Exp2 and the DCDC Mode with the jumper moved to VOT, the capacitor C25 (near the inductor of the integrated 5v power supply) explode.

after some investigation i found what i think its a problem with the EXP1 and Serial (TFT Conector), When the jumper of the NPWR is on the VOT position the TFT Conector will be supplied with the VOT power form the DCDC MODE, On the EXP1 Conector the is supplied with 5v from the integrated power supply, On the TFT24 the 5v (from NPWR) will be the same conecction as the 5v (from the 5v on the EXP1)

[image]

radek8, 2021-11-22:
You are probably right that two 5V sources will appear on the display.
But if I connect two 5V sources it should not lead to a capacitor explosion.
The cause must be another fault.
I have the same connection that you describe only with the difference TFT35.
I have no power problem.

radek8, 2021-12-15:
Did capacitor C25 really explode?
Wasn't it a C26 condenser?
Then it would look like a badly sized capacitor.
Or the wrong batch.
You are not alone to whom this happened and without an additional power supply module.
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/556

guru-florida, 2022-05-14:
I also discovered this and posted in Issue #403 .