bigtreetech/BIGTREETECH-SKR-V1.3 #553: BTT Octopus V1.1 Running TMC2209 in UART mode and cannot get my dual Z steppers and E0 to run independently

jmarkscnc, 2021-11-28:
I have been working on this issue for way to long, but I have 2- Z motors and the E0 all move at the same time? I have check everything I can think of, config.h, config_adv_h, along with the pins file along with watching about 40 hours of videos that have nothing to do with this board with no success. I have uploaded my Marlin build to my shared folder on One Drive if anyone can look it over or make some suggestion, I would be very much appreciated.

https://1drv.ms/u/s!An3HLl2sqsgkiGBMwvWDTJpdLMDs?e=rRv7Om

radek8, 2021-11-28:
Do you have a Octopus V1.1 board?
You're writing to the wrong repository.
The problem of SKR 1.3 and SKR 1.4 boards is solved here

radek8, 2021-11-28:
https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-V1.0/issues

jmarkscnc, 2021-11-28:
Closing and moved this to the correct repository.
Thank you I was doing a lot of searching and post this in the wrong place.
Moved to;
https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-V1.0/issues

closed 2021-11-28 by jmarkscnc