bigtreetech/BIGTREETECH-SKR-V1.3 #555: [SKR1.4 Turbo] Board seams dead. No USB, no Display

alabiana, 2021-12-08:
Hey Guys,

i just wanted to install my BLtouch (v3) today and updated the firmware for this with the relating changes to the configuration.h
But after placing the sdcard with the firmware.bin into the board it doesnt start anymore. I get no display signal an no uart connection (usb or tft).
The board still renames the firmware.bin to firmware.cur and even the old firmware.bin doesnt work.
Im working with platform.io and have an arduino uno here. Is there any way to unbrick the board or to activate debug or anything to see what is wrong?

JeffMedd, 2022-11-13:
@alabiana did you ever resolve this? same thing is happening to me at the moment.

alabiana, 2022-11-13:
The file was not renamed. I tried with another sd card and Compiled the Firmware again. Then it works again.

closed 2022-11-13 by alabiana