bigtreetech/BIGTREETECH-SKR-V1.3 #556: SKR 1.4 Turbo blown component

exter1312, 2021-12-10:
Hey, i changed the Creality Board on my Ender 5 Plus to a SKR 1.4 Turbo and was using it with Klipper.
It worked flawlessly until a couple of minutes ago when i saw a bright yellow flash inside the housing.

[image]

[image]

Upon further inspection i found where the flash might have come from.
I dont know what happend, since it blew mid-print.
I hope somebody can tell me what happend.

exter1312, 2021-12-10:
Ok i looked at the schematics and it seems to be C26 used for 5v DCDC circuit. I am still unsure about the reason why it happened.
[image]

radek8, 2021-12-10:
It's pretty weird.
The SKR1.4 board can supply 12-24V.
If I power it 24V then DCIN will be 24V. after voltage drop on diode V1 the voltage will be VIN = 23.5V
The VIN voltage supplies this capacitor, but according to the documentation it is only dimensioned for 10V (22UF / 10V).
It looks like a schematic error or an undersized capacitor that could explode for this reason.
I would try to desolder it, clean the surroundings, repair the connections and replace it with a new one, dimensioned at> 25V.
I have the same board and the same 24V power supply ... I think a nasty surprise awaits me :-(

[image]

[image]

radek8, 2021-12-10:
I would solve the problem with an external 5V power supply module which is supplied with this board as an optional accessory.
Remove jumpr DC SEL !!! And connected all 3 pins on connector J9
It will work and without the need for soldering.

exter1312, 2021-12-10:
Thank you for your reply.
Yes i thought about the DCDC extensionboard. Will it just replace the onboard circuit? Well replacing the capacitor will be hard since it took out the solderpads and the copper. I will give it a shot. I also contacted BTT about this issue.

radek8, 2021-12-10:
Yes, the external module replaces the on-board power supply.
However, you must completely remove the DC SEL connection in case the on-board power supply is damaged.
And connect all 3 pins together on connector J9 - NPWR. This will send 5V from the External Source to the branch of the original 5V branch.

radek8, 2021-12-10:
I would not rely too much on contacting BTT. According to user reactions, they do not communicate much. Complaints do not solve. If they answer, they will send you to the merchant from whom you bought the board .... 
:(

exter1312, 2021-12-10:
Yeah sadly i know. I bought it directly from them through aliexpress. Maybe i can get the money back though.

radek8, 2021-12-10:
Just look at the problems with the SKR2 board.
They didn't worry much about it.

exter1312, 2021-12-10:
Ok if i understood you correctly i have to bridge these three pins.
Thank you very much for your help. 

[image]

radek8, 2021-12-10:
> Yeah sadly i know. I bought it directly from them through aliexpress. Maybe i can get the money back though.

The merchant will not return the money to you alone, only Aliexperss would intervene.
And if it's been over a month, aliexpers won't deal with it

radek8, 2021-12-10:
Yes
[image]

exter1312, 2021-12-10:
If everything else fails i will try to also get paypal involved.

Thanks again. I will order the DCDC extension and keep this thread updated.

radek8, 2021-12-10:
Unplug this one completely

[image]

exter1312, 2021-12-10:
Alright. Everything should be here by Tuesday/Wednesday.

exter1312, 2021-12-14:
Update: It seems to work sofar.

radek8, 2021-12-14:
With that 5V power module?

exter1312, 2021-12-14:
Yes. I soldered the NPWR pins on J9 together and removed the DC Selector jumper.

radek8, 2021-12-14:
TMC2209  have too, they are great.

radek8, 2021-12-14:
Be careful to have the TMC 2209 switched to UART mode. Often they sell them without a jumper

exter1312, 2021-12-14:
Thanks.
I have the BTT TMC 2209 v2.0. I believe they dont have a jumper. I just removed the pin for sensorless homing.
Absolutely quiet.
Only the PSU fan is driving me crazy. Maybe i can swap it, or throttle it with resistors.

radek8, 2021-12-15:
Probably the same problem.
Maybe the wrong batch
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/552

radek8, 2022-02-12:
Please close the ticket