bigtreetech/BIGTREETECH-SKR-V1.3 #557: Delta SKR 1.4 with TMC2209s, extruder is intermittent running Marlin 2.0.9.3

rmhorwitz, 2022-01-21:
I recently got this Delta system up and running.  I noticed that some of the linear printed lines were intermittently missing.  My extruder is mechanically sound , so no substantial drag except for the smart filament detector which sometimes triggers a blocked filament error.   I placed a wrench on the extruder motor shaft and allowed it to turn while monitoring the torque by keeping a very light load on the wrench.  During the episodes of no filament extruding, I noticed that the servo motor went limp, as if power was removed momentarily.  I swapped out the TMC2209 driver, No change.  I increased the current of the servo in configuration. adv to 2000:
#if AXIS_IS_TMC(E0)
    #define E0_CURRENT      2000
    #define E0_MICROSTEPS    16
    #define E0_RSENSE         0.11
    #define E0_CHAIN_POS     -1
    //#define E0_INTERPOLATE true
The only other thing I can think of is:
#define DELTA_SEGMENTS_PER_SECOND 200

This didn't yield any improvement.  At this point, it is obviously a bug in the Marlin Software and I am open to any suggestions to resolve it. 

Any help is greatly appreciated,

Rick

radek8, 2022-01-21:
Try to increase the value
#define HOLD_MULTIPLIER
or ban it
Sets the current limit for inactive time

rmhorwitz, 2022-01-21:
Thank you.  I looked at my Pronterface log and noticed a couple entries stating `E driver over temperature warnings ( 2000mA)`
I lowered the current down to 1.6 A.   Per your recommendation I attempted to comment out the #define HOLD_MULTIPLIER, but the compiler complained so I raised it from .5 to 1.   Sadly, I still have the same problem.

radek8, 2022-01-21:
The 1.6A current on the extruder comes to me a lot.
I use 700mA and I have no problems.
In discussions, I have often seen recommendations for TMC2208 and 2209, to switch the stealthChop mode to spreadCycle for the extruder.
// #define STEALTHCHOP_E
or via terminal M569 S0 E

rmhorwitz, 2022-01-21:
Thank you again.  I am currently printing and I entered that change on the fly, but I am still getting the tell tale "Thunk!" from the extruder motor when it restarts turning.

radek8, 2022-01-21:
#define HOLD_MULTIPLIER
This choice has its meaning.
When the motor is not moving, it reduces the current flowing into the motor.
A lower current is enough to maintain the position and the motor, together with the TMC controls, does not heat up as much

radek8, 2022-01-21:
Thunk???

radek8, 2022-01-21:
Can't TMC2209 overheat?
2A is the maximum for him with very good cooling

I would try to swap TMC. If another driver will behave the same.

rmhorwitz, 2022-01-21:
Yeah, I thought the device might overheat too, so I reduced it down to 1600, but the same issue.  BTW- "Thunk!" is the sound the servo makes when it turns back on.  I think I am going to reduce the current down to 700 and see what happens.

rmhorwitz, 2022-01-21:
Oh Wow!  I read back M906 and it had the old values 2000. I must have forgot to save them from the last reset.  I set them down to 1200 and it appears to work!!!!  Redak7- you are a genius!!!  Thank you so much!!!!!

closed 2022-01-21 by rmhorwitz