bigtreetech/BIGTREETECH-SKR-V1.3 #560: SKR 1.4 Turbo cannot compile with LCD\Controller options enabled

Beochie, 2022-02-27:
Using VScode (V1.64.2) and PIO(Core5.2.5).  Marlin-bugfix-2.0.X
I cannot successfully compile if I have "#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER" enabled or any other LCD\Controller option enabled. 

In PIO, I have removed and re-added U8glib but I still cant get a good compile.  everything will compile fine until I uncomment the option mentioned above.  

I've been dinking with this for a day, I'm at a loss.  Any help would be appreciated. 

c:/users/Billybob/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: cannot open linker script file C:/Users/Billybob/.platformio/packages/framework-arduino-lpc176x/system/CMSIS/system/LPC1768.ld: Invalid argument
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\LPC1769\firmware.elf] Error 1
========================================================= [FAILED] Took 106.81 seconds =========================================================
Environment    Status    Duration
-------------  --------  ------------
LPC1769        FAILED    00:01:46.813
==================================================== 1 failed, 0 succeeded in 00:01:46.813 ==================================================== 
The terminal process "C:\Users\Billybob\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

maxholgasson, 2023-01-11:
Did you figure out a solution?

maxiwolleb, 2023-01-13:
i got the same problem

maxholgasson, 2023-01-13:
I deleted all the LCD stuff and it worked.
Not a nice solution but it would be in any case better using the latest marlin firmware..but could get it working with the SKR 1.4

maxholgasson, 2023-01-14:
> i got LCD and Printer working with Marlin 2.1.2 and a clean setup

Nice!! would you mind sharing your config files or the ones you needed to change?

maxiwolleb, 2023-01-14:
https://github.com/maxiwolleb/Marlin/tree/2.1.2-AnetA8MKS14Turbo