bigtreetech/BIGTREETECH-SKR-V1.3 #561: Which pins are i2c pins on the SKR 1.3?

Domush, 2022-03-04:
The LPC1768 has two sets of i2c pins: .0/.1 and .27/.28, but I'm not seeing where it's documented which SKR 1.3 pins these map to.

Can anyone help?

radek8, 2022-03-06:
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.3/hardware/SKR-V1.3-SCH.pdf

radek8, 2022-03-07:
[image]

Domush, 2022-03-10:
> [image]

Thank you. I've been trying to get those pins to work, but to no avail. Now I'm wondering if my i2c port is DOA, because I can't even scan for devices using those pins without the board freezing. :(

radek8, 2022-03-10:
What do you connect to the i2C port?

radek8, 2022-03-10:
I have a SKR1 board. 4. I connected an eeprom to this port and it works great.

radek8, 2022-03-20:
If that's all, can you close the ticket?