bigtreetech/BIGTREETECH-SKR-V1.3 #564: Foumelding.

Beginner69, 2022-06-21:
I wanted to upload the firmware to my Sidewinder X2 with SKR-2 board but I keep getting an error message: "Volume.init failed". How can I fix that?

Beginner69, 2022-06-25:
Dear,

thank you for the response.  I may have been a bit sloppy with my information, my apologies. 
That is my current configuration: printer is Sidewinder X1, board BTT SKR-2 (SMT32F429VGT6) , drivers TMC 2209, screen and BTT TFT3.  5 V3.  I am using a 1GB micro SD card and now insert it into the board.  Earlier I made the mistake of putting the card in the TFT screen, but I don't anymore.  I tried to make the firmware with Marlin.

radek8, 2022-06-25:
The card should be in FAT or FAT32 format.
Reformat the SD, card or use another SD card.
Some SD cards have a reading problem.
Send a video when the problem occurs.

Beginner69, 2022-06-25:
The problem for the SD is ready now. But have you a good firmware for my configuration plaese.

Vriendelijke groeten,
Guido.

Samsung Galaxy 71A

TypeApp voor Android<http://www.typeapp.com/r> downloaden

radek8, 2022-06-25:
Here you have a sample configuration.
https://github.com/MarlinFirmware/Configurations/tree/import-2.1.x/config/examples/Artillery/Sidewinder%20X2

Beginner69, 2022-06-26:
Geachte,

bedankt voor de snelle reactie.
Ik zal deze bestanden zeker van bestuderen.

TypeApp voor Android<http://www.typeapp.com/r> downloaden

https://github.com/MarlinFirmware/Configurations/tree/import-2.1.x/config/examples/Artillery/Sidewinder%20X2<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FMarlinFirmware%2FConfigurations%2Ftree%2Fimport-2.1.x%2Fconfig%2Fexamples%2FArtillery%2FSidewinder%2520X2&data=05%7C01%7C%7C83acbfc7d94f404630a408da56d18c63%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637917755244032743%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=DmziLFsRA2vr6xNopln1T76XfvVx99hWL5IGyIHzWgw%3D&reserved=0>

Beginner69, 2022-06-27:
I was looking for țhe configuration.h but wat I receive from you is not for Sidewinder X1 but for Sidewinder-X2. I am not a programmer, is it really not possible to download a good firmware for Sidewinder-X1 with a SKR-2 board and 2209 drivers?