bigtreetech/BIGTREETECH-SKR-V1.3 #565: A4988 green pcb not working, any solution?

slayernn021, 2022-07-19:
Hello world, my first post @ git :) I have read similar topic which is closed, there was no solution for problem.
I switch from ramps to skr 1.4 turbo (Marlin 2.9.0.1), got 2x tmc2099 for X,Y but i want to use my old A4988 for Z,E0 as i have ~6 of them and want to use it on this board. They are green version (chinese). So far A4988 is imposible to use, there is no motor movement, multimeter show good reading on +v pot but driver is not responding on commands. Tried different things in marlin settings(like MINIMUM_STEPPER_PULSE etc), nothing helps, its not respoding.. Green and red A4988 are pin-pin same and same controller, i see no reason why it is not working on this board and in same time it works on ramps without any problems?  Cheers!

radek8, 2022-07-20:
Are you sure you want to combine TMC2209 and A4988?
I would think twice about buying another TMC2209.
But if you're crazy about it, take both Marlin configuration files and photos of the jumer connections under the controllers.

slayernn021, 2022-07-20:
Yes i want to use A4988 for Z and E0, later in time i will switch to all-tmc2209 probably, but for now that's what i want. 
Jumpers are set correct, 4 jumpers on middle-right row (not connecting to red rows)
[Configuration_adv.md](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/9154285/Configuration_adv.md)
[Configuration.md](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/9154286/Configuration.md)
.

radek8, 2022-07-20:
About two years ago, I tried to run the TMC2208 splu with another non-TMC model.
I had the same problem. Motors with non-TMC controllers did not spin.
The problem was in the configuration. An option was enabled for TMC that caused the problem. But I don't remember which choice it was.
I see multiple TMC options enabled in your configuration.
Try disabling them and see which option causes the problem.

The first thing you should disable is
  #define STEALTHCHOP_Z
You can't set Stealthchop on A4988.

Next, I would try to disable other options. For example:
#define MONITOR_DRIVER_STATUS
#define TMC_DEBUG
#define TMC_ADV() { }

slayernn021, 2022-07-21:
> About two years ago, I tried to run the TMC2208 splu with another non-TMC model. I had the same problem. Motors with non-TMC controllers did not spin. The problem was in the configuration. An option [...]

Tried your suggestion, its not working. Problem is good known, green A4988 is not working on some (or all) skr boards.. hope someone can hack this out. Thx