bigtreetech/BIGTREETECH-SKR-V1.3 #566: TB input shorted, but can't swap it with T0/T1

lisandro52, 2022-08-28:
Hey guys, another board "fried" by a thermistor touching something.

This is what's happening and what I've tried so far:

[image]

## Facts
- The C15 (diode?) is shorted. With a multimeter, it reads almost 0 resistance (0.03 at 20k). The others read almost 5 (4.95-97).
- When I measure the inputs between them, they read 0 when reading the TB input (TH0 -> TB, TH1 -> TB). TH0 -> TH1 reads 5.
- The thermistors of the hotend and the bed measure 100 as expected.
- When I turn the machine on, both **the bed AND the hotend** read 189 ºC, even with the thermistors completely unplugged.

## What I tried
- I tried disabling the TEMP_BED_PIN. It won't show the bed's heat on the screen, and the MAXTEMP BED error won't trigger.
- Tried swapping the different PIN definitions around, but none of that worked. The error is always the same, MAXTEMP BED.

Is there something else hapenning here that doesn't allow me to simple swap the inputs around and use the others? Why is it reading 189 on all of them? Can I do something that isn't buying a new board? 

Thanks

radek8, 2022-08-31:
You can't use a pin and connect a thermistor to it to measure the temperature.
It doesn't work that way.
I have SKR V1.4 and I also managed to blow off the pin on the processor that measures the temperature of the hotend and I managed to put it on another one.
In order to replace the pin with another one, you have to find an empty pin that can handle analog voltage measurement, remap this pin in Marlin and physically connect it instead of the original pin.

What doesn't work for you?
How many hotends do you have on the printer? One or two?
If 1, you can remap the TH1 connector

radek8, 2022-09-01:
A suitable pin for remapping if you need 3 thermistors is P0_26

lisandro52, 2022-09-02:
> You can't use a pin and connect a thermistor to it to measure the temperature. It doesn't work that way. I have SKR V1.4 and I also managed to blow off the pin on the processor that measures the [...]

That's what I've tried, remapping it to TH1 but the machine keeps telling me that there is a MaxTemp error whenever I turn it on.

> Tried swapping the different PIN definitions around, but none of that worked. The error is always the same, MAXTEMP BED.

The hotend reading and the bed both say 189 ºC when I turn it on, it doesn't matter what PIN configuration I do. I have just one hotend.
I tried mapping the bed to TH0, and disabling the bed. MaxTemp error.
I tried mapping it to TH1, and disabling the others. MaxTemp Error.

That's why I'm posting here, because I think there is something else to switch the PINs around that I'm missing and that's why the FW is ignoring my changes (about the PINs, other changes do work).

For the time being, I installed everything on my old creality board and it's working (knd of, because the hotend keeps getting clogged).

radek8, 2022-09-02:
Which Pin file are you editing?

lisandro52, 2022-09-02:
I'm changing the `pins_BTT_SKR_common.h` file, and nothing else.

radek8, 2022-09-02:
This is the correct file. And what change did you make?

radek8, 2022-09-02:
What version of Marlin are you using?
According to the picture, you are using a version that is more than a year old.

lisandro52, 2022-09-02:
> This is the correct file. And what change did you make?

I was changing the PINs around in that screenshot. Tried every possible combinations I could think of: P0_25_A2 swapped with P0_23_A0 for example.

And I'm using Marlin 2.0.5.4. I started doing all this on August 2020. Everything was working fine until I pressed the thermistor too hard and all of this happened. I realized what I had done and I said 'fuck it'. Threw everything into a box and forgot about until a couple of weeks ago. That's why I'm working on an old version of the FW.

radek8, 2022-09-02:
If you set
#define TEMP_BED_PIN P0_23_A0
or
#define TEMP_BED_PIN P0_25_A2
do you get the same error? MaxTemp Error?

radek8, 2022-09-02:
Setup
#define TEMP_BED_PIN P0_26_A3
and write me if the MaxTemp Error still appears.

RandyBane478, 2023-08-11:
I know this is an older thread but I have the same problem with a Biqu B1 that has a BTT_SKR_V1.4 (not a turbo). Every time I switch to pins on the board from T0 to T1, then edit firmware in pins_BTT_SKR_common.h it gets rid of T0 reading 320/0, but it won't heat on it's own, but heats the exact amount the bed does. If any one has an idea or solution, please let me know. Thank you.

radek8, 2023-08-12:
I am also using BTT SKR V1.4.
Can you describe your problem better?
Send the configuration files you modified.

RandyBane478, 2023-08-13:
I've got it working. I found a YouTube video that referred to a closed GitHub issue that had the configuration of the pins.h file I needed. Here is the link,  https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/387.
edwilliams16 has the fix listed about halfway down in the discussion, Thank you for your quick response.