bigtreetech/BIGTREETECH-SKR-V1.3 #567: SKR 1.4 Turbo vs Tft MKS H43

andyrok, 2022-09-09:
Can someone help me I have a SKR 1.4 Turbo and I bought a Tft MKS H43 because in the specifications they gave it compatible with the SKR .. unfortunately I can't make it work ... it turns on and stays on the logo screen..😱

radek8, 2022-09-09:
Sendconfivuraci and photo involved

andyrok, 2022-09-10:
This is the suggest wiring....and the serial port...

[image]

But in Configuration_adv_h( marlin for SKR 1.4 Turbo)  "define LCD_SERIAL_PORT" is not present...
[image]

MKS wiring suggest ice this

[image]

i have try every possibility of wiring and serial port but the result is the same... only logo.....
[image]

radek8, 2022-09-10:
Did you enable configuration.h?
#define DGUS_LCD_UI_MKS

Which serial port did you connect the display to? You can't see it from your photos.
In the configuration_adv.h file, add a line with the port number.
#define LCD_SERIAL_PORT

You must not define the same port as a standard serial port in the configuration.h file

radek8, 2022-09-10:
If you do not define a serial port manually, serial port 3 will be used

[image]

andyrok, 2022-09-10:
[image]
[image]

This is the port suggest by MKS....
I use Marlin for SKR 1.4 turbo but  the line whit "LCD_SERIAL_PORT" are missed....
ok for add line "LCD_SERIAL_PORT"
I've try but only Logo appears..

radek8, 2022-09-11:
What port number did you add?

radek8, 2022-09-11:
Attention WIFI connector has only 3.3V power supply.

radek8, 2022-09-11:
Your involvement is wrong. The connection of the adapter does not correspond to the connection of the WiFi connector.

radek8, 2022-09-11:
The adapter will probably be compatible with the AUX1 connector on the SKR V1.3, but you have SKRV 1.4 and the tan does not have an AUX1 connector but WiFi.
These two connectors are not wired the same way.
Be happy if you don't destroy anything.

[image]

radek8, 2022-09-11:
[image]

[image]

andyrok, 2022-09-11:
> The adapter will probably be compatible with the AUX1 connector on the SKR V1.3, but you have SKRV 1.4 and the tan does not have an AUX1 connector but WiFi. These two connectors are not wired the same way. Be happy if you don't destroy anything.
> 
> [image]

MKS expert give me this information:

“The wifi interface is also a serial port

I thought this was right but some doubts come to me... as for the Tft power supply I think if there were only 3.3 V it wouldn't turn on...

andyrok, 2022-09-11:
[image]

radek8, 2022-09-11:
Yes, the wifi connector has a serial port, but on other pins. So you can't use a 1:1 cable

radek8, 2022-09-11:
You still haven't sent a photo of the wiring and files with the current Marlin configuration

radek8, 2022-09-11:
Make your own connection cable. 
In the evening I will send it a suggestion for his involvement.

andyrok, 2022-09-11:
[image]
I added the line: "LCD_SERIAL_PORT"

[image]

[image]

[image]
I created the link with single cables... As in the scheme....

radek8, 2022-09-11:
If you want to use the WiFi connector, you must define serial port 3
#define LCD_SERIAL_PORT 3

The display requires a 5V power supply. use the 5V pin on the TFT connector.

Did you uncomment this line?
#define DGUS_LCD_UI_MKS

Comment
#define serial_port_2 1

radek8, 2022-09-11:
If you would like to define serial port 0,
you must not connect the power supply to the WiFi connector, but you would have to use the TFT connector on the SKR

andyrok, 2022-09-11:
ok, but the line #define DGUS_LCD_UI_MKS is not present
[image]

radek8, 2022-09-11:
[image]

andyrok, 2022-09-12:
Ok, so I have to compile Marlin ... not using the specific firmware of the SKR 1.4 Turbo...?
Dwin set folder I already installed it in the H43.... I put the microsd and the Tft update was successful...
Your help is very important ... thank you!!