bigtreetech/BIGTREETECH-SKR-V1.3 #568: Replacing LPC1768 CPU

papcorp, 2022-09-27:
Hello, I was cleaning the nozzle/heatblock after some print failure with a copper brush, i know this is stupid but have done it a milion times before so i got lazy and confident.
either way,  shorted the heater wire with thermistor wire and puff goes the skr 1.3 board. i did notice the cpu and coil directly below it was fried, ordered new CPU and replaced it along with burnt coil. tried to upload a firmware to the board via SD card but no success (the file is the same, doesnt change to CUR). when i turn on the printer (ender3 with stock screen) the screen continously screams and shows nothing, leds on board are lit and what is strange heatbed seems to be warming up. 
so the questions:
1. is there a possibility that something else is burnt on the board, and any hints where i should look for?
2. or is there a possibility bootloader is not enabled by default with 'from factory' cpu and I have to enable it somehow
3. if 2 is yes, how to do it?
4. any other suggestions?

radek8, 2022-09-28:
Yes, the new processor does not include a bootloader.
First you have to load the bootloader in order to be able to load the new FW via the SD disk

radek8, 2022-09-28:
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/346

radek8, 2022-09-28:
https://github.com/GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader