bigtreetech/BIGTREETECH-SKR-V1.3 #569: marlin not compiling when I change TEMP_SENSOR_BED

brunatort1101, 2022-11-02:
Hello, I just started compiling marlin for a SKR 1.4 turbo, and after some mistakes and learning finally managed to install marlin on the board. Everything works fine except the hotbed temperature reading, which is always 0 and the beb won't heat. Went to marlin and found that TEMP_SENSOR_BED was set at 0, changed it to 1 and marlin won't compile. it returns a "region RAM overflowed with stack
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\LPC1769\firmware.elf] Error 1"
I spent all the day searching a way to solve this problem. I even tested the solution comented in this post: https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/387   but had no luck.
I put here mi configs, I hope somebody can find what is going wrong
[configuration adv h.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/9922217/configuration.adv.h.txt)
[configuration h.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/9922218/configuration.h.txt)

radek8, 2022-11-03:
Why are you compiling such an old version of Merli?
Why don't you compile the latest versions?

brunatort1101, 2022-11-03:
> Why are you compiling such an old version of Merli? Why don't you compile the latest versions?

I tried with the lastest with the same luck, it just happens that this version is the las one I tried and pick it up. I don't know what is happening to cause that error only when I select the bed sensor

brunatort1101, 2022-11-04:
Sooo..., it seems that the flash memory was really filled up to the roof. Tried to disable extra features and the problem dissapeared when I disabled SD_SUPPORT, no, now the board read the bed temp, it heats ok anr everything seems to work ok. coming from ramps I though that sdsupport was necessary to read a car from the screen, but it is only to be able to print from the board itself. since I am ok with using the screen as the card reader, this problem is solved. I hope this can help others

closed 2022-11-04 by brunatort1101