bigtreetech/BIGTREETECH-SKR-V1.3 #571: SKR 1.4 Turbo not visible on pc via usb

Navally, 2023-01-27:
Hi there!
I bought an SKR 1.4 Turbo to my Anet ET4+ and it worked perfectly but when i mount ESP-01S to that MB and flashed marlin with configuration Serial port 1: 3, Serial port 2: -1 then i cant connect it with my PC via USB cable. I tried to flash older configuration with Serial port 1: -1 and Serial port 2: 0 it still don't connect with PC.
Any idea where is the problem?
Thanks!

radek8, 2023-01-28:
The way you configured it is fine.
option -1 activates the USB port
option 0 activates the serial port of the TFT connector
option 3 activates the serial port of the WIFI connector

Can you send the configuration files?

Navally, 2023-01-28:
[Marlin.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/10528395/Marlin.zip)

radek8, 2023-01-28:
The ports are defined correctly.
But why are you compiling an old version of Marlin?

Navally, 2023-01-28:
I had this version, and latest from marlin github. On both versions is the same with USB connection.

radek8, 2023-01-29:
I have the same board, + TFT display + ESP-01S
and the USB port works for me.

radek8, 2023-01-29:
After connecting the printer to the PC, check whether you have a problem with the drivers on the PC

Navally, 2023-01-29:
Drivers are ok, on second PC is the same.

Navally, 2023-01-31:
Thanks for your time! I think there is a problem with chip lpc1769 and i contact with the seller and he send to me second MB.

closed 2023-01-31 by Navally