bigtreetech/BIGTREETECH-SKR-V1.3 #573: skr 1.4 turbo - thermistors always in open state, ADC out of range

KnYaZuK, 2023-03-05:
Hi there! A have an issue with the thermistors connection.
I v already tried marlin 2.1.2, 2.0.9.5 and some older versions.
Also a tried latest version of the klipper firmware.
Everething works fine, but not the thermistors. I think the problem in mainboard. I buy mainboard in the middle of 2019 and it was installed only now in 2023 =)
Where should i look? The adc debug shows 4095 value. The pronterface and my btt  tft display shws me zero celcius. The thermistors i use is 104gt-2. I am already bought new one - the result always one - 4095 adc value.
The Klipper firmware also crashes witth "ADC out of range error"
With the marlin- if i warm up the thermistor to the 60-70 celcius - it will show me the temperature and the ADC value will droped down to the lower then 4000 values.
Accordingly i made a conclusion - the problem in the mainboard.

I am also have the Trigorilla 1.1 and on them thermistors works fine and shows me the correct ambient temperature.

Please, community, help =\ I tired...

radek8, 2023-03-06:
Do you compile your own Marlin?

KnYaZuK, 2023-03-06:
> Do you compile your own Marlin?

Yes, i also tried some pre configured builds and result the same. For now i ordered the new skr 1.4 board. Not turbo. But i want to understand what happened wrong with my skr 1.4 turbo board.
Maybe some of us already faced with the same problem. I was not found any similar cases here and inside the marlin repository issues.

The `t1`, `t0` and `tb` pins have the 3.3 volts when the board powered from the 12v power supply.
The thermistors has the correct resistance and it are changed when warming\cooling.
If i shorted any of `t` pins - the marlin firmware catches this case correct and throws the `MAXTEMP triggered` error

radek8, 2023-03-06:
Send the configuration files.

KnYaZuK, 2023-03-06:
@radek8 
https://github.com/KnYaZuK/Marlin/tree/kossel-212
https://github.com/KnYaZuK/Marlin/tree/kossel-2095

KnYaZuK, 2023-03-11:
Just a little update. The new skr 1.4 (not turbo) board has arrived. I have installed the Klipper firmware with default config for skr 1.4 and changed the thermistor to the apropriate type `ATC Semitec 104GT-2` and everything works fine. The temperature shows the ambient value.