bigtreetech/BIGTREETECH-SKR-V1.3 #577: SKR 1.4 TMC2209 TURBO Configuration files for CR-10s Broken

joshlacr43156, 2023-04-05:
Hello everyone, 

This is my first time posting attempting to get help so if there is any information I have left out I am very sorry and will happily provide it. 

I am currently swapping out my old creality v1.2 motherboard for the SKR 1.4 TMC 2209 TURBO in my Creality CR-10s and have hit a major road block trying to compile the firmware. 
I have changed the "default_envs =  LPC1769"
When compiled VS code gives this error (full error log at the end): 

(user)\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\bits\move.h:33:10: fatal error: bits/c++config.h: No such file or directory

Things I have tried and the results:

I tried deleting "toolchain-gccarmnoneeabi" and recompiling, which resulted in it successfully downloading and reinstalling toolchain-gccarmnoneeabi, and then giving the same no such file or directory error. 

I have tried bugfix version 2.1 of marlin, 2.0, even tried an earlier version and they all give the same error. 

I tried compiling with configuration files for a different board and compiled perfectly, leading me to believe there is a problem with the configuration files for this board for CR-10s.

I am at a complete loss and would be incredibly grateful for any help or guidance. Thank you in advance!

Here is the complete terminal output:

PS C:\Users\joshw\Downloads\Marlin-bugfix-2.1.x\Marlin-bugfix-2.1.x> platformio run --silent -e LPC1769
In file included from c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\bits\stl_pair.h:59,
                 from c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\utility:70,
                 from c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\array:38,
                 from C:\Users\joshw\.platformio\packages\framework-arduino-lpc176x\cores\arduino/HardwareSerial.h:26,
                 from Marlin\src\HAL\LPC1768\MarlinSerial.h:24,
                 from Marlin\src\HAL\LPC1768\MarlinSerial.cpp:24:
c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\bits\move.h:33:10: fatal error: bits/c++config.h: No such file or directory
   33 | #include <bits/c++config.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\MarlinSerial.cpp.o] Error 1
In file included from c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\bits\stl_pair.h:59,
                 from c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\utility:70,  
                 from c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\algorithm:60,
                 from Marlin\src\HAL\LPC1768\../../inc/../HAL/./LPC1768/HAL.h:33,
                 from Marlin\src\HAL\LPC1768\../../inc/../HAL/HAL.h:30,
                 from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\LPC1768\HAL.cpp:24:
c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\bits\move.h:33:10: fatal error: bits/c++config.h: No such file or directory        
   33 | #include <bits/c++config.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\bits\stl_pair.h:59,
                 from c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\utility:70,
                 from c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\algorithm:60,
                 from Marlin\src\HAL\LPC1768\../../inc/../HAL/./LPC1768/HAL.h:33,
                 from Marlin\src\HAL\LPC1768\../../inc/../HAL/HAL.h:30,
                 from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\LPC1768\HAL_SPI.cpp:51:
c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\bits\move.h:33:10: fatal error: bits/c++config.h: No such file or directory        
   33 | #include <bits/c++config.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\HAL.cpp.o] Error 1
In file included from c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\bits\stl_pair.h:59,
                 from c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\utility:70,
                 from c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\algorithm:60,
                 from Marlin\src\HAL\LPC1768\../../inc/../HAL/./LPC1768/HAL.h:33,
                 from Marlin\src\HAL\LPC1768\../../inc/../HAL/HAL.h:30,
                 from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\LPC1768\MinSerial.cpp:24:
c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\bits\move.h:33:10: fatal error: bits/c++config.h: No such file or directory        
   33 | #include <bits/c++config.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\HAL_SPI.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\MinSerial.cpp.o] Error 1
[... 52 lines cut ...]
                 from C:\Users\joshw\.platformio\packages\framework-arduino-lpc176x\cores\arduino/CDCSerial.h:28,
                 from Marlin\src\HAL\LPC1768\main.cpp:31:
c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\cstdlib:41:10: fatal error: bits/c++config.h: No such file or directory
   41 | #include <bits/c++config.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\main.cpp.o] Error 1
In file included from c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\bits\stl_pair.h:59,
                 from c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\utility:70,
                 from c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\algorithm:60,
                 from Marlin\src\HAL\LPC1768\tft\../../../inc/../HAL/HAL.h:30,
                 from Marlin\src\HAL\LPC1768\tft\../../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\LPC1768\tft\tft_spi.cpp:23:
c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\bits\move.h:33:10: fatal error: bits/c++config.h: No such file or directory        
   33 | #include <bits/c++config.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\tft\tft_spi.cpp.o] Error 1
========================================================================== [FAILED] Took 3.71 seconds ========================================================================== 

Environment    Status    Duration
-------------  --------  ------------
LPC1769        FAILED    00:00:03.714
===================================================================== 1 failed, 0 succeeded in 00:00:03.714 =====================================================================PS C:\Users\joshw\Downloads\Marlin-bugfix-2.1.x\Marlin-bugfix-2.1.x> echo "done" >"C:\Users\joshw\AppData\Local\Temp\ipc"
PS C:\Users\joshw\Downloads\Marlin-bugfix-2.1.x\Marlin-bugfix-2.1.x>

TMillesich, 2023-04-06:
Okay, this might be far-fetched, but i can find the file "c++config.h" 33 times inside the download of the xpack-arm-none-eabi-gcc. 
https://xpack.github.io/dev-tools/arm-none-eabi-gcc/releases/

It has always the same content!!!

How about you copy that one from \c++\11.3.1\arm-none-eabi\bits to  \c++\11.3.1\bits ?

joshlacr43156, 2023-04-09:
@TMillesich  Thank you very much for your suggestion. Following this idea I copied the "c++config.h" from 

C:\Users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\arm-none-eabi\bits

to 

C:\Users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\bits

Which did solve the particular error I was getting before, but then spit out out an error about another file. So I repeated the steps and added in the next missing file. This, again, solved the issue but spit out another missing file. At this point I simply copied all of the files from the original folder and pasted them into the secondary folder. 

This again, solved the issue of missing files, but has now resulted in a new error stated below. I feel like this has made serious progress though, and am hopeful. Thanks in advance for any help you may be able to offer. 

Processing LPC1769 (board: nxp_lpc1769; platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.3.zip; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1769.html
PLATFORM: NXP Arduino LPC176x (0.1.3) > NXP LPC1769
HARDWARE: LPC1769 120MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES:
 - framework-arduino-lpc176x @ 0.2.9
 - toolchain-gccarmnoneeabi @ 1.110301.0 (11.3.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 1 compatible libraries
Scanning dependencies...
Dependency Graph        
|-- Servo @ 1.0.0       
Building in release mode
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\HAL.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\HAL_SPI.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\MarlinSerial.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\MinSerial.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_flash.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_sdcard.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_wired.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\fast_pwm.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\include\digipot_mcp4451_I2C_routines.c.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\include\i2c_util.c.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\main.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\tft\tft_spi.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\tft\xpt2046.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\timers.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\u8g\LCD_I2C_routines.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\u8g\LCD_pin_routines.c.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_hw_spi.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_st7920_hw_spi.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_st7920_sw_spi.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_sw_spi.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\usb_serial.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\Delay.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\HAL.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\MinSerial.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\backtrace\backtrace.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\backtrace\unwarm.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\backtrace\unwarm_arm.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\backtrace\unwarm_thumb.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\backtrace\unwarmbytab.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\backtrace\unwarmmem.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\backtrace\unwinder.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\backtrace\unwmemaccess.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\eeprom_api.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\esp_wifi.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\servo.cpp.o
Compiling .pio\build\LPC1769\src\src\MarlinCore.cpp.o
Compiling .pio\build\LPC1769\src\src\core\serial.cpp.o
Compiling .pio\build\LPC1769\src\src\core\utility.cpp.o
Compiling .pio\build\LPC1769\src\src\gcode\calibrate\G28.cpp.o
Compiling .pio\build\LPC1769\src\src\gcode\config\M200-M205.cpp.o
[... 117 lines cut ...]
Compiling .pio\build\LPC1769\FrameworkArduino\HardwareSerial3.cpp.o
Compiling .pio\build\LPC1769\FrameworkArduino\Print.cpp.o
Compiling .pio\build\LPC1769\FrameworkArduino\SoftwareSPI.cpp.o
Compiling .pio\build\LPC1769\FrameworkArduino\SoftwareSerial.cpp.o
Compiling .pio\build\LPC1769\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\LPC1769\FrameworkArduino\Tone.cpp.o
Compiling .pio\build\LPC1769\FrameworkArduino\WInterrupts.cpp.o
Compiling .pio\build\LPC1769\FrameworkArduino\Wire.cpp.o
Archiving .pio\build\LPC1769\liblpc176x.a
Compiling .pio\build\LPC1769\FrameworkArduino\arduino.cpp.o
Compiling .pio\build\LPC1769\FrameworkArduino\main.cpp.o
Archiving .pio\build\LPC1769\libCMSIS.a
Archiving .pio\build\LPC1769\libFrameworkArduino.a
[... cut ...]
c:/users/joshw/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld.exe: cannot find crt0.o: No such file or directory
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\LPC1769\firmware.elf] Error 1
================================================== [FAILED] Took 22.79 seconds ==================================================
Environment    Status    Duration
-------------  --------  ------------
LPC1769        FAILED    00:00:22.792
============================================= 1 failed, 0 succeeded in 00:00:22.792 =============================================

joshlacr43156, 2023-04-11:
Anyone still having issues with this like I was, there is a solution! Or at least one that worked for me. Thank you very much to everyone that helped with this, and special thanks to @radek8 for the solution.

To fix the problem I downloaded the most current stable version of marlin (2.0.9 for me) and ignored any instructions of copying and pasting configuration and configuration.adv files. Just modify the all ready existing configuration files in the marlin download. Its a bit more effort but works. I used the creatlity 10s configuration files to get information on things like bed size, jerk settings, etc etc but do not copy and paste the files themselves.

I also found these videos very helpful:

https://youtu.be/-Gdk0wHg51w
https://youtu.be/oHMZ0ocTYvM
https://youtu.be/j2FL_mY_LsY

closed 2023-04-11 by joshlacr43156