bigtreetech/BIGTREETECH-SKR-V1.3 #578: SKR 1.4 Turbo Compiling Broken

joshlacr43156, 2023-04-10:
Hi there everyone, 

I am trying to compile my new skr 1.4 turbo with marlin using VS Code and I am running into an error I can not solve. Opening the firmware provided in VS Code and attempting to compile it gives the following error:

Warning! Ignore unknown configuration option `monitor_flags` in section [env]
Warning! `src_filter` configuration option in section [env:DUE] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:SAMD51_grandcentral_m4] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:STM32F4] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:STM32F7] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:malyan_M300] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:esp32] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy31] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy35] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_build_flags` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use `build_src_flags` instead
Warning! `src_filter` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use 
`build_src_filter` instead
Warning! `src_filter` configuration option in section [env:include_tree] is deprecated and will be removed in the next release! Please use 
`build_src_filter` instead
In file included from Marlin\src\gcode\stats\../../inc/../HAL/./LPC1768/../shared/Marduino.h:36,
                 from Marlin\src\gcode\stats\../../inc/../HAL/./LPC1768/HAL.h:39,
                 from Marlin\src\gcode\stats\../../inc/../HAL/HAL.h:26,
                 from Marlin\src\gcode\stats\../../inc/MarlinConfig.h:30,
                 from Marlin\src\gcode\stats\../gcode.h:294,
                 from Marlin\src\gcode\stats\M31.cpp:23:
Marlin\src\gcode\stats\../../module/../libs/duration_t.h: In static member function 'static void GcodeSuite::M31()':
Marlin\src\gcode\stats\../../module/../libs/duration_t.h:130:35: warning: 'sprintf' may write a terminating nul past the end of the destination [-Wformat-overflow=]
  130 |     if (y) sprintf_P(buffer, PSTR("%iy %id %ih %im %is"), y, d, h, m, s);
C:\Users\joshw\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Arduino.h:58:18: note: in definition of macro 'PSTR'
   58 | #define PSTR(v) (v)
      |                  ^
In file included from Marlin\src\gcode\stats\../../module/printcounter.h:25,
                 from Marlin\src\gcode\stats\M31.cpp:25:
Marlin\src\gcode\stats\../../module/../libs/duration_t.h:130:21: note: 'sprintf' output between 15 and 22 bytes into a destination of size 
21
  130 |     if (y) sprintf_P(buffer, PSTR("%iy %id %ih %im %is"), y, d, h, m, s);
In file included from c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\functional:56,
                 from c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\pstl\glue_algorithm_defs.h:13,
                 from c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\algorithm:74,
                 from Marlin\src\lcd\../inc/../HAL/./LPC1768/HAL.h:35,
                 from Marlin\src\lcd\../inc/../HAL/HAL.h:26,
                 from Marlin\src\lcd\../inc/MarlinConfig.h:30,
                 from Marlin\src\lcd\lcdprint.h:35,
                 from Marlin\src\lcd\lcdprint.cpp:31:
c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\bits\functional_hash.h:169:3: error: redefinition of 'struct std::hash<long unsigned int>'
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\bits\functional_hash.h:137:3: note: previous 
definition of 'struct std::hash<long unsigned int>'
  137 |   _Cxx_hashtable_define_trivial_hash(wchar_t)
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
*** [.pio\build\LPC1769\src\src\lcd\lcdprint.cpp.o] Error 1
======================================================= [FAILED] Took 21.70 seconds =======================================================

Environment    Status    Duration
-------------  --------  ------------
LPC1769        FAILED    00:00:21.701
================================================== 1 failed, 0 succeeded in 00:00:21.701 ==================================================PS C:\Users\joshw\Desktop\Marlin-2.0.x-SKR-V1.4-Turbo> echo "done" >"C:\Users\joshw\AppData\Local\Temp\ipc"
PS C:\Users\joshw\Desktop\Marlin-2.0.x-SKR-V1.4-Turbo> 

Any tips or suggestions as to how to get this working would be greatly appreciated! Thank you for your help

radek8, 2023-04-11:
You have to use the version of Marlin that you download from github and write the settings into it.
You must not copy configuration files from another branch into the downloaded Marlin.

joshlacr43156, 2023-04-11:
> You have to use the version of Marlin that you download from github and write the settings into it. You must not copy configuration files from another branch into the downloaded Marlin.

@radek8 thank you very much for this suggestion. I will try it right now. In your opinion, should I use a stable version of Marlin from the marling github page? Or should I use a stable version provided by BigTreeTech github page?

joshlacr43156, 2023-04-11:
> You have to use the version of Marlin that you download from github and write the settings into it. You must not copy configuration files from another branch into the downloaded Marlin.

@radek8  SUCCESS!!!!! Thank you so much, I could have sworn I had tried this all ready but I believe I had previously tried editing the configuration files for creality V1.

To be clear for anyone else fighting this problem: Something is wrong with the configuration files provided for cr10s. 

To fix the problem I downloaded the most current stable version of marlin (2.0.9 for me) and ignored any instructions of copying and pasting configuration and configuration.adv files. Just modify the all ready existing configuration files in the marlin download. Its a bit more effort but works. I used the creatlity 10s configuration files to get information on things like bed size, jerk settings, etc etc but do not copy and paste the files themselves. 

I also found these videos very helpful:

https://youtu.be/-Gdk0wHg51w
https://youtu.be/oHMZ0ocTYvM
https://youtu.be/j2FL_mY_LsY

closed 2023-04-11 by joshlacr43156